labrad/servers #18: HTTP stuff

pomalley, 2015-01-07:
@joshmutus -- what's the status of the http stuff? It doesn't have to be completed but it would be good to consolidate it down to one branch.

joshmutus, 2015-01-07:
> Assigned #18 <https://github.com/martinisgroup/servers/issues/18> to

Yeah, I think I can just merge the branches...

zchen088, 2015-01-29:
Is this in reference to the HTTP server?

zchen088, 2015-01-30:
If it's in reference to the HTTP servers, it appears that it's hard coded to look for the Vince cryo log, and the pressure display was hard coded to look for the "test hack" version of the server. It could also use an overhaul, since there's things like the diode temperatures not being named. I'm wondering if we can have it pull data from the DR logger instead? That way we might be able to plot it or something fancy like that

If this is not about the HTTP server, I can make a new issue for this.

pomalley, 2015-01-30:
This is actually about a different HTTP server that's still (?) under development, a replacement for the browser.

For the current HTTP server, yes, it should take from either the DR logger or the datavault. Good catch.