labrad/servers #21: ADR GUI scrolling up / move it to servers / add me to pyle

andrewdunsworth, 2015-01-09:
Peter.

DanielSank, 2015-01-09:
Shouldn't this be three issues?

andrewdunsworth, 2015-01-09:
Yes, it should, but they're all for me (peter) to do so I told him to do this.

--peter on andrew's account

andrewdunsworth, 2015-01-20:
Did you happen to fix the auto scrolling of the check boxes?

pomalley, 2015-01-20:
No, not yet, but if you want to move the ADR GUI folder over you can follow these instructions:

http://gbayer.com/development/moving-files-from-one-git-repository-to-another-preserving-history/

Might as well do the fridge guardian too, do you agree, @DanielSank?

zchen088, 2015-01-20:
How are we differentiating which GUIs are in pyle and which are in servers?

pomalley, 2015-01-20:
My thought was that the ones that essentially provide an interface to a server go in servers, and ones that are more related to data taking go in pyle.

zchen088, 2015-01-20:
I don't think we have any GUIs that are related to data taking right now. Should we just move all of them to servers?

pomalley, 2015-01-20:
Maybe the Wiring Editor?

commit faa01b38 (2015-03-27): Change the ADR GUI to only optionally hide unviewed lines.

Also, handle time being returned as DimensionlessFloat
Review: maffoo, fixes #21.
PR #60 (2015-03-27): ADR GUI checkbox hiding
There is now an option to hide checkboxes for un-watched lines, rather than showing them on mouseover (which turned out to be annoying). This fixes #21, finally.
closed 2015-03-27 by pomalley