labrad/servers #22: VNA Server has a VISA timeout error when the number of VNA averages is too large

andrewdunsworth, 2015-01-10:
It might also have to do with the sweep settings (such as number of points, and IF bandwidth).  

Here is the error report:

[image]

pomalley, 2015-01-10:
1) What server function causes this?
2) In that function, what are the GPIB commands being sent?

andrewdunsworth, 2015-01-10:
I'm pretty sure this has to do with sending data to the datavault actually.

andrewdunsworth, 2015-01-10:
I think it is where it is trying to save it hangs and the timeout is just a symptom of this.

closed 2015-01-10 by andrewdunsworth

andrewdunsworth, 2015-01-10:
A problem seems to occur in a call to pyle.plotting.dstools.getDataset

pomalley, 2015-01-10:
I'm not sure how this causes a VISA timeout, but if you can't load a dataset then that is certainly a problem.

andrewdunsworth, 2015-01-10:
It might be unrelated....

andrewdunsworth, 2015-01-10:
Currently it seems you can get away with just doing one datavault average and this is my kludgey workaround.

pomalley, 2015-01-10:
OK. I'm still not quite buying the fact that a data vault thing is causing a VISA timeout but if you have a workaround for now... I'll sit down with you at some point and we can figure this out.