labrad/servers #25: Lakeshore Upgrade?

andrewdunsworth, 2015-01-19:
When filling cryogens it would be nice if we could toggle the lakeshore to read out more rapidly on the mix plate ruox and not on the other channels so we can watch the rise times of things on shorter time scales.

pomalley, 2015-01-19:
I don't know why the current timing was chosen, but it's possible that we may not be able to go faster if

* it's limited by the settling time of the measurement (more likely, especially as this is a low current measurement)
* switching heats stuff up (less likely)

andrewdunsworth, 2015-01-19:
But is there a way we could just measure on the same channel for these times instead of switching? Like can we toggle between switching and not switching, that would be enough of an improvement for it to be worth it to me

pomalley, 2015-01-19:
Probably there is, yes. :-)

ejeffrey, 2015-01-19:
It should already be configured to alternate the mix with everything 
else.  So the measurement order is:

1,2,1,3,1,4,1,5,...

I think the limit isn't the switching time, but the integration time of 
the ADC.  Whether we can speed that up or not, I don't know.  Probably 
yes given that we don't really need all 6 digits of resolution...

Evan

pomalley, 2015-01-19:
Evan is correct, as usual. It would be pretty straightforward to add a setting to the server to switch the read order between "1,2,1,3..." and "1". But he's also right that I don't know how much that would gain.

andrewdunsworth, 2015-01-20:
Even a factor of 2 would be useful.

pomalley, 2015-01-20:
It looks like it may even be possible to stop the cycling now, without changing the code.

The `selectchannel` function (https://github.com/martinisgroup/servers/blob/master/lakeshore370.py#L512) sets the `onlyChannel` variable, and if this is greater than zero then it only reads that channel (see here: https://github.com/martinisgroup/servers/blob/master/lakeshore370.py#L343). (Looks like the channels are 1-indexed on the hardware end of things.) So you should be able to do:

```python
ls = cxn.lakeshore_ruox()
ls.select_device()
ls.select_channel(1)  # now you're not switching, presumably channel 1 is the mix.
ls.select_channel(0)  # now you are
```

So give this a shot and see if it makes a difference.

Next step would be to try tweaking the settling time (https://github.com/martinisgroup/servers/blob/master/lakeshore370.py#L114), preferably adding a setting so you can change it on the fly.

If these work then we can talk about how to add this functionality to the GUI.

pomalley, 2015-01-20:
If we want to change the settling time there are some settings we can change on the hardware end of things, too, like the resistance range or the voltage applied and stuff like that.

andrewdunsworth, 2015-01-20:
So awesome.

andrewdunsworth, 2015-01-20:
Why is the settling time hard coded?

pomalley, 2015-01-20:
Historical reasons. That is, I dunno.

ejeffrey, 2015-01-21:
Do not change the excitation voltage.  Anything else is fine, but that has
a real possibility to heat up the fridge.

Evan