labrad/servers #33: Values should be converted to DAC units in the FPGA server, not the DAC calibration server.

pomalley, 2015-02-11:
AFAICT, the deconvolution server is what converts the full scale range (-1, 1) to the DAC range (0, 2**14).

This line: https://github.com/martinisgroup/servers/blob/master/ghzdac/correction.py#L852

This should be done in the GHz FPGA server. It would be nice to e.g. be able to plot the corrected data.

DanielSank, 2015-02-11:
While I agree that the deconvolution server should not be dacifying the wave forms, I'm not convinced that the fpga server should be doing it either. If the FPGA server is to do it I would say that it should be done in the various `DAC` subclasses to make sure that different numbers of DAC bits are supported.

DanielSank, 2015-02-23:
I see that things are moving here. Do you want me to put the dacification into the DAC subclasses? I'm fine to do this as long as everyone agrees that it makes sense.

pomalley, 2015-02-23:
I'm sort of half-way through a qubit sequencer refactoring-ish thing (@maffoo helped a lot but I've been slow). We also have the DAC builds now, too. So let's talk this out; I'll be in in a minute.

DanielSank, 2015-02-23:
I'm at the Google building but we can chat after J-Dog's thesis offense.

JulianSKelly, 2016-03-31:
Given that I have completed my thesis defense, I think we should proceed on this front.

DanielSank, 2016-03-31:
Yeah good point. @pomalley, get your butt over here.