labrad/servers #34: Qubit sequencer should support receiving deconvolved fourier data.

pomalley, 2015-02-11:
Currently the only way to give pre-deconvolved data to the qubit sequencer is if you give it time domain data. If we're going to do deconvolutions in pyle, it should support receiving deconvolved frequency domain data as well.

maffoo, 2015-02-11:
There's really no point in keeping frequency domain data around once you've corrected for the response. I'd say that the deconvolution should just fourier transform back to the time domain as well, just as the correction code currently does. Then it wouldn't be necessary to modify the qubit sequencer at all.

pomalley, 2015-02-11:
You're right, I was confused.

However, the data returned from the calibration server are in DAC units. The qubit sequencer does not have the ability to receive DAC'ified data. (For deconvolve=false the qubit sequencer DAC'ify's the data itself; see here: https://github.com/martinisgroup/servers/blob/master/Qubit%20Server/dev/src/org/labrad/qubits/channeldata/IqDataTime.java#L25)

So this ties in with #33--where do we want to DACify the data?

maffoo, 2015-02-11:
Given that the qubit sequencer can distinguish between floating point data and integer data, it should be easy to keep the current behavior (complex data that is already deconvolved is rescaled to DAC units assuming +/-1 full scale) and add the ability to handle dacified data (integer data is passed along as-is to ghz dacs server).

pomalley, 2015-02-11:
Yes, it proved pretty easy (though this is untested): a4bab60e19046977c0ad5fa955fcc67872137b7f

ejeffrey, 2015-02-11:
I think ideally the fpga server should be responsible for knowing the DAC
resolution and the packing format and doing dacify.  We would need to send
the ECL data separately then.

Evan

DanielSank, 2015-02-11:
So you want to send +1/-1 scaled floats for I and Q to the fpga server.

pomalley, 2015-02-11:
That would be my preference, but I'm not sure exactly how this works with the dynamic range that's set aside by the calibration server.

ejeffrey, 2015-02-11:
Yes, that is my preference.  Either the qubit sequencer or the
deconvolution server should make sure we have spare dynamic range as
necessary (so that a piAmp of 1 -> 0.5 to the fpga server), but doesn't
care whether '1' is 2^13 or 2^7 or something else, and certainly doesn't
know that the high 2 bits of each 16 bit word correspond to ECL outputs.

Evan

DanielSank, 2015-02-12:
I agree with this. Back in ancient history I separated out the constants for the different FPGA boards/builds. That info now lives in `DAC.py` and `ADC.py`. As an immediate fix we can put the bit resolution there. After that I'd like to break that info into non-python files so we have a once-and-for-all resource which can be read from different programming languages.

pomalley, 2015-02-12:
So we're in agreement then. Since Julian and I are knee-deep in this at the moment, we'll try to figure out how to best do this as we deploy the jump table stuff.

commit fc38f1f2 (2015-02-12): Allow qubit sequencer to receive DAC'ified data.

The settings "SRAM IQ Data" and "SRAM Analog Data" now have the option
to accept (*i{I}, *i{Q}) and *i, respectively, for SRAM data. In this
case deconvolve must be false (i.e. the deconvolution has already been
done). The data are to be in DAC units, i.e. integers from 0 to 0x3fff.

The binary (.jar file) has also been updated.

Fixes #34.

Reviewed by maffoo.
closed 2015-02-12 by pomalley