labrad/servers #43: Qubit Server Refactoring--memory commands and jump table commands

pomalley, 2015-02-17:
Most commands issued to the qubit sequencer are issued to a particular channel. The memory commands, however, are a bit different because they are handled inside `FpgaModelDac`, but sometimes issued to a particular channel (which forwards the command to its FpgaModelDac instance), and sometimes issued to the whole experiment (which sends a command to each FpgaModelDac).

The jump table commands, on the other hand, are specific to a channel, with the one exception that for analog boards, the two channels must have identical jump tables. `FpgaModelDac` has two subclasses, `FpgaModelMicrowave` and `FpgaModelAnalog`.

Since we want to support both old and new boards, I'm trying to figure out the best way of refactoring the FpgaModel structure.

Current ideas are:
1. We have `FpgaMicrowaveMemory`, `FpgaMicrowaveJumpTable`, `FpgaAnalogMemory`, `FpgaAnalogJumpTable`, which I don't like because however we do the inheritance we'll be duplicating either the memory/JT handling code or the microwave/analog code.
2. Mixins seem like they might be the solution, but Java doesn't do mixins.
3. Instead we just use object composition, i.e. give the `FpgaModelDac` base class a "Controller" object that would be either `MemoryController` or `JumpTableController` and put the relevant functions in there.

I think the problem is that (1) memory<->jump table is not a one-for-one mapping because generally the memory sequences are the same for all the boards, but not necessarily for the jump tables (but maybe they should be) and (2) I'm probably overthinking this.

Basically I was hoping @maffoo would state the obvious answer I'm missing.

maffoo, 2015-02-17:
Memory sequences are not identical for all boards, but they have the same structure. For example, when you add a memory command to set fast bias levels on some number of channels, the qubit server will create memory sequences with the send commands on the boards that control those bias channels, and idles everywhere else, so you end up with memory sequences that proceed in lockstep.

I think we probably want to handle the jump table in the same way. You specify the jump table program to the qubit server once for the entire experiment, and the server then builds the actual jump table commands for each individual board. All boards end up running the same program in lockstep. I don't see a reason to allow different jump table programs on different boards, it seems that is needlessly complicated unless we have a clear use case in mind.

As for the implementation, I think option 3 is definitely preferable. Even if java had mixins, I think object composition would be easier to understand.

pomalley, 2015-02-24:
OK, I did this as we talked about. See [here](https://github.com/martinisgroup/servers/tree/jump_table_bench_test/Qubit%20Server/dev/src/org/labrad/qubits/controller), as well as QubitContext, FpgaModelDac, etc.

closed 2015-02-28 by pomalley