labrad/servers #47: interpol1d_cubic_S in correction.py is not used

DanielSank, 2015-02-20:
It should be nuked.

closed 2015-02-20 by DanielSank

maffoo, 2015-02-20:
Just wanted to point out here that for small stuff like this, there's probably no need for a separate issue. The pull request gets assigned an issue number and can have discussion about the change etc. Essentially a pull request is an issue where the code to resolve it is already done and included right there in the issue. 

So, the idea would be if the change is small and you're working on it anyway, just make the pull request. If you're not working on it immediately, or want someone else to take it on, or it's a bigger change and you want to gauge interest or get feedback before investing lots of effort, then a separate issue is probably worth it.

DanielSank, 2015-02-20:
Indeed. I think @pomalley actually made an issue about this before and we decided to work precisely as you're saying here. In fact, I think this is even discussed in the contributing file. I was just on autopilot.

PR #48 (2015-02-20): Remove unused function interpol1d_cubic_S.
Fixes #47 .