labrad/servers #52: Time out setup state in GHz FPGA server

ejeffrey, 2015-03-09:
Currently, the GHZ FPGA server keeps track of a hardware state, and decides whether to run the setup packets based on whether the current scan is using the same settings as the previous ones.  This avoids resending packets to slow devices such as the GPIB bus on every sequence.  However, it has no way to tell if I manually change the settings from the front panel of a signal generator.

What it should do is time-out that state after (say) 60 seconds idle.  Thus, as long as a scan is running, the hardware configuration is changed only when necessary.  We assume that users are not changing the GPIB devices during a scan.  However, if the system is idle, we assume someone might have made manual changes and we reprogram the hardware before starting up again.  Since this will happen at most 1/minute, it will have negligible impact on performance.

Assigning @DanielSank because he is familiar with the GHZ FPGA server, but this would be a good small project for an FPGA server padawan to get familiar with that rather complicated piece of code.

maffoo, 2015-03-10:
Timeout is a good, simple option, but it won't catch everything (if someone makes a changes faster than 60 seconds, or whatever the timeout is) and it will also slow down subsequent scans in the common case where in fact nothing has changed but the setup state has timed out. This may not be too bad if the only setup state is uwave generators and fastbias cards, but if we add microwave switches or other setup that requires a wait time before running (#55), then this could start to become more prohibitive. Is there any way that we can get notified when an instrument gets put into manual mode? If so, we could monitor that and notify the fpga server explicitly to drop the setup state.

ejeffrey, 2015-03-10:
I have looked, and as far as I can tell there is no reliable way to tell if
a device is put in local mode

The Radiall switches have a switch time of 15 ms, so it isn't really
prohibitive.  Even if you give it an extra 100 ms settling time it isn't
too big a deal.  The only caveat is that if we put the cold switch under
automatic control we need to be really careful to not switch it when
unnecessary to avoid heating the fridge.  I think this can be handled in a
way specific to the switches.

Another approach here would be to make a variant setup packet that is an
assertion -- you assert that the switch must be in the correct position and
it either succeeds or fails.

Evan

maffoo, 2015-03-10:
Yeah, I was more worried about needing to wait for the fridge to cool down after switching the cold switch; if you're thinking of room-temperature switches then I agree the times you mention are acceptably short.