labrad/servers #53: Session level pipelining for qubit sequencer

ejeffrey, 2015-03-09:
Right now, we allow sequences from multiple scans to be interleved via the same system we use for pipelining requests from a single scan.

This is really flexible, but somewhat inefficient, as the fastbias and anritsus must be programmed every time we switch between scans.  So I propose adding a more coarse grained exclusion mechanism in the qubit server.  The idea is that each scan would request a 'sequence group ID' token.  The qubit sequencer would then run only sequences from that group until it completes, or for up to a set period of time on the order of a couple of minutes.  Then the next scan would run for a couple of minutes, and so forth.

The upshot of this would be that short scans (rabihigh) will just run in order with no interleaving.  However, a running long scan can be preempted by a short scan, which will run to completion and then resume the long scan.  Long scans will be interleved, but on longer intervals to avoid the overhead of switching the hardware configuration.

This could be enhanced by adding priority hints like "low priority / long scan", or "high priority / short scan" or "uninterruptable scan" (for things like the ramsey oscilloscope that need precise timing", but I think the initial implementation can make some reasonable defaults and have it 'just work'

What do people think?  Any suggestions?

pomalley, 2015-03-10:
As we discussed this in person, you already know I think this is a good idea. :-)

For the sequence token, do you envision having a call to the qubit sequencer in the scan, outside of func? Or in `sweeps.grid`, etc? (The problem with `sweeps` is that things like adaptive scans don't use it.) What about when a scan doesn't provide a token--assign them the "default token"?

I noticed you self-assigned this--does that mean you want to start working on the qubit sequencer? I'll provide a bit of friendly advice: don't let anybody know you know Java, or they'll come to you with their sequencer problems!

Seriously though, if you do want to implement this, it should be pretty straightforward at the most basic level. Here's my first guess:
1. Create a `SequenceToken` class with static functions to get / check tokens. The `Resources` class provides a basic example of how to do this concurrently (i.e. with the `synchronized` keyword). I guess there will be some static variables to actually keep track of the tokens, as well as who's currently running.
2. Create a LabRAD setting in `QubitContext` to get a new token, as well as one to set the token for the current context.
3. Add some code in `run_experiment` to query the `SequenceToken` code for permission to run.
4. (Make sure you put the `build_sequence` part outside of the run lock so that you don't have to wait your turn just to get e.g. a configuration error.)

ejeffrey, 2015-03-10:
I was going to automatically assign sequence tokens by default based on the
connection ID with a short default timeout (15 seconds or so).  So all
sequences run from a single connection would be assumed to be in the same
group.  This alone I think would get us 90% of the way there with no
changes in pyle.  Then I would add new settings to create a new sequence
token, select a specified sequence token within a context, and to
explicitly yield when the sequence finishes.  These could be called from
sweeps.grid or sweeps.run.

Adaptive scans still use sweeps.run.  The bigger problem is with nested
scans, you probably only want to set the sequence key at the top level.