labrad/servers #54: Fastbias serial support for qubit sequencer

ejeffrey, 2015-03-10:
Qubit sequencer should support programming fastbias over serial connection.

zchen088, 2015-03-10:
Evan, I thought you already got this to work?

pomalley, 2015-03-10:
@zchen088: I believe the fastbias works over serial but there is no channel for it in the qubit sequencer.

The natural way to do it for the qubit sequencer is to write another channel class. The problem is that the `Channel` interface specifies `getFpgaModel`/`setFpgaModel` methods which don't apply here. I guess the simple solution is to bump those down a level into `SramChannel` (and `FastBiasChannel`), and if we want to get really fancy, make `FpgaModel` a subclass of some sort of `HardwareModel` and have `FastBiasSerialModel` or something also a subclass of that.

N.B. There exists the `FastBiasChannel` class which is for FastBias control through a DAC board, so don't be confused about that. Also, there are `PreampChannel` and `TriggerChannel` still floating around and I don't know that we ever use those anymore.

pomalley, 2015-03-10:
I'm going to do this now, but it will be in the jump table branch.

pomalley, 2015-03-11:
This commit (https://github.com/martinisgroup/servers/commit/e5975f11bf454fabfeed6768cfc34a6db52abc54) has a first pass at implementing this. @maffoo, let me know if you think I'm headed in the right direction here.

Still todo: there is no hardware information from the DC rack server in the qubit sequencer. There should be some sort of resource model for this, at least to check that the card you specify for your FastBiasSerialChannel actually exists before you send the setup packet.

ejeffrey, 2015-03-11:
> This commit (e5975f1

Hi Peter,

Why not keep the same channel type name and FastBiasChannelBuilder, and
have the channel builder figure out if it is a serial or FPGA controlled
channel by reading the wiring key?  Does this somehow not work because of
the way the memory sequences work?

pomalley, 2015-03-11:
Actually, you're probably right, but the way to determine it would be to look at the first parameter and see if it's a DAC name (e.g. "Vince DAC 1") or a DC rack card number. (And then check that against the wiring key to see if it is valid.)

pyle/fpgaseq will have to keep track of whether to call mem_bias or serial_bias, though. (maybe there's a way to use inheritance to its fullest and use the same command but it seems icky)

maffoo, 2015-03-11:
I think the direction here looks good. I agree that the information about whether fast bias boards are controlled by serial or fibers should be in the wiring info and figured out for you. Also, I think the command to set the levels should be called "Config Bias Voltage", to go along with all the other "Config *" settings that configure properties of the various channels that do not vary in time during the sequence.

ejeffrey, 2015-03-11:
I think we only want to do that if we can use the same bias command.  The
idea would be that I could change the wiring key and automatically switch
to using the serial port. The problem is that mem_bias has functionality
that the serial port can't emulate: changing the bias voltage multiple
times in a sequence.

So I think the way this would work is that there would be a new
config_bias_voltage setting that would either A) create a setup packet for
the serial port or B) add a mem_bias command depending on the wiring key.
 mem_bias would then raise an exception if you attempted to call it on a
fastbias channel that was attached to a serial port.  We then change
fpgaseqtransmon to call the new config_bias_voltage instead of mem_bias.

Evan

pomalley, 2015-03-11:
This makes sense. I'll need to look at the actual arguments to mem_bias because I think they're somewhat different and see if it actually makes sense to make a mem_bias call out of a simple (channel, voltage) call to config_bias_voltage. Maybe this will require config_bias_voltage to set the biases for all channels, for example (which is probably fine).

Also, I haven't exactly, uh, tested any of this yet, so there's that, too.

pomalley, 2015-03-11:
The next pass is in: back to only one type of builder and renamed the setting to `config_bias_voltage`.

Still TODO:
1. Call mem_bias from config_bias_voltage if appropriate, else config the serial bias channels.
2. Check wiring for valid DC Rack card (i.e. include that in wiring at all).

pomalley, 2015-03-11:
It's commit 92b04c61b8163b4b80dd2ab0c47de7f1c50b89dd.

pomalley, 2015-04-29:
We have shown this to be working on Vince, at least when the serial server is working.

closed 2015-04-30 by pomalley