labrad/servers #55: Switch channel type for qubit sequencer

ejeffrey, 2015-03-10:
Qubit sequencer should support a "switch" resource for microwave switches.

pomalley, 2015-03-10:
Is the use case for this simply to make sure to switch to your experiment before running the sequence? Or rather, could we enumerate the functionality that this would provide so that we can think about the best way to implement it?

Another option, for example, would be to treat it similarly to how we do microwave sources. There's a `MicrowaveSource` class and a corresponding `MicrowaveSourceConfiguration` class, and you call `Config Microwaves` to create a configuration, and when it builds the sequence it checks that the configurations are compatible. Now, the configurations are owned by the various microwave channels, which is how you figure out what hardware goes with what configuration, but I imagine something similar could be done for the switch.

Aside: "switch" is also a verb so I was very confused by the title until I actually read the description. :-)

maffoo, 2015-03-10:
I would agree with handling this like we do microwaves, since it's a static configuration for the experiment like the microwave source frequency and power, not a time-varying thing like dac or adc channels. Also like microwaves, we could have wiring configuration that specifies how the switches are connected; then we might be able to determine the switch settings automatically, and detect if you try to run an experiment that would require conflicting switch settinfgs. The switch config should be sent as "setup packets" to the fpga server, like we do with the microwave packets. We probably also want to build in some delay between switching and actually running an experiment; how long do you typically wait after switching to let things settle?

ejeffrey, 2015-03-10:
Yes, the idea is to make sure that the switch(es) are in the right position
before you start your experiment.  It would be handled through setup
packets.

I am not sure what you mean by "do it like the microwave sources".  I see
two options: add a new channel type, and add a new setting: "setSwitch(
(device_name, channel_name), switch_position)" or tie it to an existing
DAC/ADC channel and have the the qubit sequencer figure out which switch is
connected to which DAC based on the wiring key.

I don't really like the latter option as it is somewhat limiting.  There is
no reason that switches have to be tied to a FPGA channel.  For instance,
we have this problem with the paramp pump: you can't set the microwave
configuration on an anritsu that isn't hooked up to a DAC for e.g., the
paramp pump.

maffoo, 2015-03-10:
I think a hybrid approach could work. We include switches in the wiring and any uwave channels that you want to use for your experiment will get automatically set up by the qubit sequencer for you with setup packets. For any other switches, we could add a new setting like you suggest. Note that the qubit sequencer already lets you add arbitrary setup packets with the 'config_setup_packets' setting, but we could add one which is more specific to the way we want the switches to operate so that users don't have to construct the setup packets manually.

pomalley, 2015-04-29:
This one is not making it in the JT PR, not sure why I added it to the milestone.