labrad/servers #58: What is the 'Vince' branch and why is it the active branch on Trench?

DanielSank, 2015-03-11:
I am very confused by this. Can someone who was doing experiments in Trench (in the trenches?) before the APS meeting give me a clue?

DanielSank, 2015-03-27:
Badump.

maffoo, 2015-03-27:
for the newbie, what is trench?

DanielSank, 2015-03-27:
LabRAD managing computer on dry DR.

maffoo, 2015-03-27:
ah, well I see a lot of @pomalley commits in the log for the branch that don't seem to be on master; maybe he can shed some light?

pomalley, 2015-03-27:
Actually, those commits by me are cherry-picked from the old branches where I was working on the DR logger and SuperMonitor stuff. They should be in master at this point, I believe.

pomalley, 2015-03-27:
It's the other commits I'm not sure about.

jwenner, 2016-02-25:
The active branch on trench has been master for some time now.

closed 2016-02-25 by jwenner