labrad/servers #59: FPGA server swallowing setup packets errors

pomalley, 2015-03-11:
For some reason the exception raised here:

https://github.com/martinisgroup/servers/blob/master/GHzDACs/ghz_fpga_server.py#L815

is not getting propagated back to the user and instead things hang. I don't know why, but if I put in a print statement before I can see the message.

pomalley, 2015-03-28:
Ugh, I got hit by this again today. I'd like to restate how friggin terrible this is.

maffoo, 2015-03-29:
Can you tell which packet send is failing? It looks `sendAll` is invoked in three places, for [Load packets](src/main/scala/org/labrad/qubits/controller/MemoryController.scala), [Setup packets](https://github.com/martinisgroup/servers/blob/master/GHzDACs/ghz_fpga_server.py#L665), and [Read packets](https://github.com/martinisgroup/servers/blob/master/GHzDACs/ghz_fpga_server.py#L665). The case of Setup packets, the deferred returned by `sendAll` is being immediately yielded, but in the case of load and read, there are locking calls (runLock.acquire and readLock.release) happening between the call to `sendAll` and actually yielding to capture the results (or thrown exception). So, if either of those locking operations fails, then the exception could be swallowed. If we could determine which invoking was hanging, it could help narrow down what's happening.

pomalley, 2015-03-29:
Both times so far it was the setup packets--specifically the setup packets for the DC rack server.

pomalley, 2015-05-16:
I believe this has been fixed with the changes to error handling in the FPGA server.

closed 2015-05-16 by pomalley