labrad/servers #62: Serial server not able to talk to MKS gauges on Vince

zchen088, 2015-03-14:
At some point on Thursday, MKS gauge server lost contact with the hardware. @JulianSKelly might know more about when/how this happened.

In debugging, we tried writing and reading via the NI MAX software, which worked and gave us the correct pressures. We were also successful using pyvisa. However, we couldn't seem to communicate with the gauges using pyserial. Does anyone know how pyserial and pyvisa interact with the gauges differently?

pomalley, 2015-03-14:
I have updated the serial server and the DR logger in the branch
fixing-dr-logger. Try using that if you weren't already; I changed the
timeout from 0 to 0.01. Don't know why I needed to do that, though.

JulianSKelly, 2015-03-14:
Cool that seemed to fix my problem. Me being Jimmy.

pomalley, 2015-03-14:
Boom! Paternity leave, schmaternity leave.

DanielSank, 2015-03-14:
How come so many people in the lab look like Julian?

pomalley, 2015-03-14:
Also it would be awesome if someone could see about getting the
fix-dr-logger branch merged in, because I'm not sure when I'll be able to
get to it. It should tie in with the newer supermonitor to give you lots of
info about how the dr logger is failing (if it is).

pomalley, 2015-06-08:
The MKS gauges on Vince are working now.

closed 2015-06-08 by pomalley