labrad/servers #63: MKS gagues not always automatically detected

JulianSKelly, 2015-03-18:
Supermonitor on Vince currently running with only 'still' and ''return' pressures.

pomalley, 2015-03-18:
Are you running the MKS server (i.e. not the testhack or whatever)? If the
MKS server's get_gauge_list returns the gauges, the dr logger should pick
it up.

JulianSKelly, 2015-03-18:
They don't seem to show up:

<code><pre>
In [1]: cxn.mks_gauge_server.get_gauge_list()
Out[1]: ['Still', 'Return']
</pre></code>

Thoughts?

ejeffrey, 2015-03-19:
If the USB->serial adapters got moved to a different port, they will show
up as the wrong COM port, and the server won't be able to find them.  Even
if you didn't move them, windows might have just renumbered the COM ports.
Check the device manager against the registry.

JulianSKelly, 2015-03-19:
The ports are 4, 5, 6, 7, 8, 9 in device manager
The gauge ports are 5, 6, 7, 8 in mks_gauge_servers/vince/gauges

So, as far as I can tell this is not the problem.

JulianSKelly, 2015-03-19:
@zchen088 unplugged and replugged something and everything is fine now.

Jimmy, what exactly did you do?

zchen088, 2015-03-19:
It appears the problem was that something had opened the COM ports tied to the gauges so that they couldn't be used by the MKS gauge server. I figured this was the case because I couldn't open the COM ports via the serial server, even with the MKS gauge server closed. The easiest way I knew to reset the COM ports was to unplug the USB and plug it back in, which worked.

Does anyone know of a way to force all COM ports to close their connections without touching the hardware? @pomalley

pomalley, 2015-03-19:
Not offhand. Unplugging and resetting the serial server may be the way to
go, for now. Ideally the dr logger should now handle it gracefully.

zchen088, 2015-03-19:
I think we still had to restart the DR logger for all the gauges to show up again on the SuperMonitor.

pomalley, 2015-03-19:
You know what you're right. It only looks up the gauges on first connect.
Maybe this is something to add in, but of course you'd have to create a new
dataset, and I don't know how the supermonitor would handle a new dataset
coming in with different columns than the first.

closed 2022-04-22 by JulianSKelly