labrad/servers #64: GHz_fpga_server times out

JulianSKelly, 2015-03-19:
<code><pre>
2015-03-18 16:38:10-0700 [ServerProtocol,client] GHz FPGAs now serving
2015-03-18 16:54:32-0700 [ServerProtocol,client] Traceback (most recent call las
t):
2015-03-18 16:54:32-0700 [ServerProtocol,client] Failure: labrad.types.Error: (0
) Operation timed out [payload=None]
2015-03-18 16:54:32-0700 [ServerProtocol,client] RECOVERING FROM TIMEOUT
2015-03-18 16:54:33-0700 [ServerProtocol,client] 'executionCounter'
</code></pre>

Leaves buffered packets in direct ethernet server

maffoo, 2015-03-19:
Some more info would be helpful here:
- are those logs from the node server?
- how often do you see this?
- is it reproducible (say on particular scans or in some other way)?
- does this leave the system in a broken state? if so, what do you have to do to recover?
- is this a new failure or something you've seen before?

DanielSank, 2015-03-19:
This is only a half joke: get another ethernet switch and see if the problem goes away. This has actually been the solution we found after days of debugging *several* times in the past.

> is this a new failure or something you've seen before

This problem comes around about once a year. At least twice in the past it was never understood but solved by trashing the ethernet switch and putting in a new one.

* What firmware are the DACs on?
* Does this happen if you run only one board? Two? Three?

JulianSKelly, 2015-03-19:
*   That log is from running the server directly in command line
*   It has been crashing every ~10 minutes
*   I haven't seen this for a few years
*   You recover the system by restarting the ghz_fpga_server, and then the qubit sequencer
*   The DACs are all build 8
*   Its not reproducible in terms of a particular scan
*   This happens when running single qubit scans, at least.

I'll try playing with the 10 mhz reference cables. 

Do we have extra switches in lab?

DanielSank, 2015-03-19:
Try running with one *board*. To do this I think you just set `config` to have only one qubit in it.

JulianSKelly, 2015-03-19:
This is strongly correlated with DAC 15 losing phase lock. Every time we get a crash, DAC 15 has unlocked. We're going to try and fix that problem, then see if the timing out goes away.

DanielSank, 2015-03-19:
In case you guys haven't fixed this let me just brain dump what I would do:

* Tighten the clock cable for the suspicious board.
* If that doesn't do it, switch that board to a different clock line to find out if it's the clock channel or the board.
* If it's the board replace it.
* If it's the line, try a different cable and put the 10 MHz reference out onto the scope to inspect visually.
* Review all ground connections and look for anything strange.
* Switch suspicious board to different power supply channel.

maffoo, 2015-05-20:
bump. do we still have an issue here @JulianSKelly?

DanielSank, 2015-05-20:
This was from a `KeyError

JulianSKelly, 2015-05-20:
I haven't seen this error in some time.

closed 2015-05-20 by DanielSank