labrad/servers #67: Can't run ghz_fpga_server

jwenner, 2015-03-21:
When Charles and I tried to run ghz_fpga_server from a command line (master branch), it won't start. In particular, I get the following message:
```
U:\Jim\Code\servers\GHzDACs>ghz_fpga_server.py
Traceback (most recent call last):
  File "U:\Jim\Code\servers\GHzDACs\ghz_fpga_server.py", line 222, in <module>
    import servers.GHzDACs.Cleanup.dac as dac
ImportError: No module named servers.GHzDACs.Cleanup.dac
```
I then tried running Cleanup\dac.py (which indeed is present!):
```
U:\Jim\Code\servers\GHzDACs\Cleanup>dac.py
Traceback (most recent call last):
  File "U:\Jim\Code\servers\GHzDACs\Cleanup\dac.py", line 3, in <module>
    import servers.GHzDACs.Cleanup.fpga as fpga
ImportError: No module named servers.GHzDACs.Cleanup.fpga
```
Upon trying to run fpga.py:
```
U:\Jim\Code\servers\GHzDACs\Cleanup>fpga.py
Traceback (most recent call last):
  File "U:\Jim\Code\servers\GHzDACs\Cleanup\fpga.py", line 8, in <module>
    from servers.GHzDACs.util import LoggingPacket
ImportError: No module named servers.GHzDACs.util
```
However, ```U:\Jim\Code\servers\GHzDACs>util.py``` does run successfully.

The only way we (actually, Rami) could figure out how to get past this was to move ghz_fpga_server.py to the folder containing servers (here, ```U:\Jim\Code\```), however, it then complains:
```
U:\Jim\Code>ghz_fpga_server.py
Traceback (most recent call last):
  File "U:\Jim\Code\ghz_fpga_server.py", line 226, in <module>
    from util import TimedLock, LoggingPacket
ImportError: No module named util
```

Is there any way to get ghz_fpga_server to actually run??!!??

(PS: Please keep Charles in the loop as well.)

ejeffrey, 2015-03-21:
1) The directory containing servers has to be on the PYTHONPATH
2) Don't change the system python path on the datataking computer to point
to your U:\ checkout.  If you need to run servers from the U drive, please
make a shortcut/batch file that sets the python path as you need it.
3) You shouldn't need or want to run servers from the master branch from
the U drive.  If you are developing the fpga server, you should be working
in a branch, if not, you should be using the version on the manager host
and started by the browser/node. If it won't run on the manager make sure
to note that.

jwenner, 2015-03-21:
Evan, thank you for the response. It now makes sense why it didn't run (not having servers on PYTHONPATH). The problem was that ADR computer in Pedram's lab doesn't even have a git checkout of servers (but it has an SVN checkout), so I was trying to run from my own checkout. I've forwarded your reply to Charles (but won't close the issue until know that it's resolved).

DanielSank, 2015-03-21:
For now the right way to do this is make a directory somewhere on the machine called whatever you want, `labrad-servers` for example. Inside that directory clone the git repo for the servers. Set the PYTHONPATH environment variable to `labrad-servers`.

jwenner, 2015-03-23:
Charles was able to get this to work (although he told me that he had to place servers within pylabrad for PYTHONPATH to work). Thanks!

closed 2015-03-23 by jwenner