labrad/servers #74: Serial issues with MKS gauges

pomalley, 2015-03-25:
The MKS server on vince periodically loses its brains and the way to fix it is to unplug the USB-serial devices and plug them in again and then restart the serial server and then the MKS gauge server. Fortunately the DR logger seems to handle this somewhat gracefully now but it's still not ideal. I don't know what is causing this; presumably the COM ports get locked or scrambled somehow.

This is probably related to #68, #63, #62 and maybe some more stuff too.

JulianSKelly, 2015-04-06:
Had this happen again today. I being Jimmy. Could this be related to some timeout setting somewhere? The last time we definitely fixed a problem witht he mks_server, it had to do with timeout.

pomalley, 2015-04-10:
Redheat bluescreened at some point last night. I tracked it down to the USB->Serial driver. So I installed a newer (I think) version of the driver. (See bugify for details.) Maybe this will fix it? Or it could be a hardware problem.

zchen088, 2015-04-15:
This really really needs to get fixed. Can someone work with me on this tomorrow afternoon?

DanielSank, 2015-04-15:
I will try.

pomalley, 2015-04-15:
@christopherquintana reverted the serial server on mrfreeze back to the one that's currently running on trench. could be that timeout=0 wrong, and things weren't working on vince for... other reasons? (e.g. hardware or driver problems, one piece of evidence in support of this is the redheat bluescreen.)

zchen088, 2015-04-19:
More info: Trench (the dry DR control computer) has had working serial connections without issue. It is running both and older version of python (2.6) and older version of pyserial (1.31 vs. 2.7 on all the computers having this problem).

zchen088, 2015-04-19:
I implemented a total hack on redheat: Installed python 2.6 and pyserial 1.31, then ran the serial server from the command line, along with setting the timeout back to 0. Things appear to have worked for the last 5 minutes, but we'll see if it's still working tomorrow.

With the timeout set to 1 and in this hacky python 2.6 configuration, I did get the serial server process to go into it's zombie state again where I had to manually power cycle the computer to get it to go away. The last thing it printed was "deferredRead timeout at 2.0 seconds" or something like that. Does this mean anything to anyone? @pomalley @maffoo

zchen088, 2015-04-19:
It turns out when serial.VERSION says 1.31, it is actually pyserial 2.2, for some stupid reason. I will try again when I get the chance with python 2.7 and see if reverting to pyserial 2.2 works.

DanielSank, 2015-04-19:
maybe `serial.VERSION` is a protocol version?

zchen088, 2015-04-19:
Confusingly serial.VERSION on pyserial 2.7 does indeed give 2.7. Also serial doesn't have a __version__ attribute.

pomalley, 2015-06-08:
This seems to be fixed now.

closed 2015-06-08 by pomalley