labrad/servers #75: fpga.pll_init() returns labrad type error

rpbarnes, 2015-03-26:
I'm currently using build 14 with GHzDAC_V8_B14_P22_P45_s15.sof loaded onto the fpga. I'm running python 2.7.9 on windows 7.

I use the 'jump_tables_bench_test branch' of servers repo.
I use up-to-date version of pylabrad, pulled from repository.
I run labrad, direct ethernet, and ghz_fpga_server.py (from jtbt branch). Screen shot of running programs below.

[image]

I then run the following code to initialize the fpga and get the type error.

[image]

I run the same lines from above in ipython and get a more detailed traceback.

[image]

I insert a print statement in ghz_fpga_server.py and run the code in ScreenShot2 and get a response on the server end, shown in screen shot below along with code to generate print statement.

[image]

Please let me know if you need anymore detail or if you're having trouble generating the error.

pomalley, 2015-03-27:
The jump table stuff is still under heavy development. Appreciate the bug report, hope to fix it soon!

rpbarnes, 2015-03-27:
Thanks! I've been conversing with Daniel as well. Do let me know if you 
need any help reproducing the error.
~Ryan

DanielSank, 2015-03-28:
I don't think this is a jump table problem per se. It looks like something more basic broke in going to the updated code though.

pomalley, 2015-04-07:
This should be fixed by 96cc31784068. Not sure when we'll reconsolidate all the JT branches, let alone get it into master.

pomalley, 2015-05-16:
This is now merged to master.

closed 2015-05-16 by pomalley