labrad/servers #77: FPGA server hanging on collectPkts

pomalley, 2015-03-28:
(jump table branch)

The FPGA server is not returning from `yield`ing the DeferredList of collectPkts. As I understand it, these packets should make the direct ethernet server wait for the appropriate number of response packets or timeout. (I believe only the ADC should be sending back packets.) There are no packets coming back, and it does not appear to be timing out.

I don't really know how to debug this further. Thoughts?

DanielSank, 2015-03-28:
Can you check the timeout set when the collect call is made to the direct ethernet server?

pomalley, 2015-03-28:
Yes, the timeout is 600001, which certainly explains why it's not timing out. Perhaps there's a units issue...

pomalley, 2015-03-28:
Indeed, see #49--I just replaced it with 100*reps, thinking that 100 microseconds was a reasonable temporary hack. Turns out it's interpreted in seconds. So I guess this counts as either "units error" or, more realistically, "Peter doesn't know how the FPGA server actually works."

pomalley, 2015-03-28:
Still doesn't explain why there are no packets coming back, though, especially when I got it to work last time.

DanielSank, 2015-03-28:
Glad you figured out the timeout. The reason this problem happened is that although @ejeffrey and I improved the way pylabrad deals with the `v` tag, the Delphi API still uses the crappy old way. I hesitate to bother fixing it since @maffoo already rewrote the manager and direct ethernet duders in scala.

As for why the packets aren't coming back: check to see if they're being read by the direct ethernet server. If they are then we have a problem in the fpga server. If they are not then either 

* There's something wrong with the ethernet communication, such as the switch randomly deciding to be a PITA, which happens.
* There's something wrong with the daisychain/triggering of the boards such that they aren't running as many times as they're supposed to. You can figure out a lot of this by looking at the packets received by the direct ethernet. If you're still stuck we can apply fpgaTest to the problem. I will try to keep checking back here.

pomalley, 2015-03-29:
Thanks, I appreciate the help.

Fixing the labeling issue in #78 now helps with the debugging. Packets are coming back to the DE server but they don't seem to be correct at all.

```
2015-03-28 17:04:54-0700 [ServerProtocol,client] RECOVERING FROM TIMEOUT
DEBUG:root:acquired lock
DEBUG:root:no setup
DEBUG:root:bothPkt sent
DEBUG:root:acquired readLock
DEBUG:root:released runLock
DEBUG:root:collecting results...
ERROR:root:TimeoutError: Some boards failed:
Vince DAC 29: OK Expected executions: 600  Actual: 600
Vince DAC 17: OK Expected executions: 600  Actual: 2673
Vince ADC 1: timeout! Expected executions: 600  Actual: 33280
2015-03-28 17:04:54-0700 [ServerProtocol,client] attempt 5 - error: Some boards
failed:
2015-03-28 17:04:54-0700 [ServerProtocol,client] Vince DAC 29: OK Expected execu
tions: 600  Actual: 600
2015-03-28 17:04:54-0700 [ServerProtocol,client] Vince DAC 17: OK Expected execu
tions: 600  Actual: 2673
2015-03-28 17:04:54-0700 [ServerProtocol,client] Vince ADC 1: timeout! Expected
executions: 600  Actual: 33280
2015-03-28 17:04:54-0700 [ServerProtocol,client] FAIL!
DEBUG:root:collected results
DEBUG:root:pageLocks released
2015-03-28 17:04:56-0700 [ServerProtocol,client] Traceback (most recent call las
t):
2015-03-28 17:04:56-0700 [ServerProtocol,client] Failure: labrad.types.Error: (0
) Operation timed out [payload=None]
ERROR:root:Recovering from timeout...
2015-03-28 17:04:56-0700 [ServerProtocol,client] RECOVERING FROM TIMEOUT
ERROR:root:TimeoutError: Some boards failed:
Vince DAC 29: OK Expected executions: 600  Actual: 600
Vince DAC 17: OK Expected executions: 600  Actual: 51500
Vince ADC 1: timeout! Expected executions: 600  Actual: 22272
2015-03-28 17:04:56-0700 [ServerProtocol,client] attempt 5 - error: Some boards
failed:
2015-03-28 17:04:56-0700 [ServerProtocol,client] Vince DAC 29: OK Expected execu
tions: 600  Actual: 600
2015-03-28 17:04:56-0700 [ServerProtocol,client] Vince DAC 17: OK Expected execu
tions: 600  Actual: 51500
2015-03-28 17:04:56-0700 [ServerProtocol,client] Vince ADC 1: timeout! Expected
executions: 600  Actual: 22272
2015-03-28 17:04:56-0700 [ServerProtocol,client] FAIL!
```

I have no idea where those numbers are coming from...

pomalley, 2015-03-29:
So I'm now thinking that this may be a result of unlatched PLLs. The ADC and one of the DACs both return True for pll_query. I think I'm getting the same bug as #75 for pll_init on the DAC.

DanielSank, 2015-03-29:
Take a look at the light to the left of the FPGA as you look at the board from the front of the shelf. Do you see a light on? That light means PLL unlatch and is fixed by the bringup script (if things are working). Wasn't there an issue recently from @JulianSKelly about unlocks in which I advised a review of the clock cables, etc? Since I guess you're on Vince I think this is probably at least related. Could this be a problem with the new firmware, as you noted by the reference to #75? (I'm just spewing information here).

pomalley, 2015-03-29:
I believe the light is on for the ADC but not the DAC (on the same plate). I will look at the cables.

pomalley, 2015-03-29:
OK, I can't actually tell what's what and I've got to go. I'll pick this up again later.

DanielSank, 2015-03-29:
Note that the ADC has some additional lights which indicate overload of the input. These should flicker on/off in normal operation because of noise. We should document what these lights are better. There's a lab TWiki page for the fpga boards. Let's use it!

ejeffrey, 2015-03-29:
Wait, that sounds wrong...   The ADC input is "supposed" to overload in
normal operation?

DanielSank, 2015-03-29:
Well, yeah. We operate at reasonably low SNR and that's after summing over ~70 time samples. So, one expects that if I choose a signal amplitude which takes up a respectable fraction of the ADC dynamic range, the Gaussian noise should probably be railing the ADC now and then. Right?

If this sounds high to you I could go calculate this carefully.

ejeffrey, 2015-03-29:
A single point clipping now and then is probably not a problem, but if it
is regularly hitting the rails we will be losing SNR -- if the ADC is at
the rail, the signal is zero.  Furthermore, if it is clipping while idle, I
am worried it will clip more while measuring.  What is the maximum
amplitude when reading out 5 qubits compared to the RMS noise amplitude?

It may be tricky to tune the gain in this well with 8 bit ADCs, but it
should be possible to avoid clipping.  As long as the white noise component
is much larger than 1/2 click on the ADC, your quantization noise is
negligible, and decorrelated.  That should be all that is necessary to get
maximum SNR.  So if the white noise has an RMS amplitude of 16 clicks (more
than enough to dither the signal and swamp the quantization noise), we
would be 8 standard deviations away from clipping on the 127 maximum
amplitude.  If the noise is gaussian, it would clip about once per week at
1 GS/s.  If we decrease the RMS amplitude to 10, it would essentially never
clip.

I would be interested to know how the clipping indicators work.  Do they
have pulse stretchers so that even a single clipped point causes a visible
flicker?  Or does it mean that if you can see the light flickering there is
a lot of clipping?

So one thing I think might have confused this is that we recently replaced
the diffamps with higher gain versions, and we saw an increased SNR.
However, these amplifiers also had *dramatically* lower noise due to the
design.  It is possible that the extra gain is not necessary for the ADC,
and that we would be better off with a low-gain but low-noise version of
the diffamps.

Probably the first thing to do here is to check with a scope and see how
much it is actually clipping compared to how much flickering we see on the
LED.

Evan

DanielSank, 2015-03-29:
> So one thing I think might have confused this is that we recently replaced
the diffamps with higher gain versions, and we saw an increased SNR.

What?!?!?!? That means we're throwing away SNR for being retarded.

ejeffrey, 2015-03-30:
It really depends on how much we are clipping.  It could easily be a
trivial amount.  It could be a source of intermodulation distortion.  It
could cause a reduced SNR.  I don't really have very good intuition for
what happens when an ADC clips and the total noise is much greater than the
signal power.

Evan

DanielSank, 2015-03-30:
Oh, I think I misunderstood something you said before. When we increased the gain of the diff amps and got better SNR, are you talking about a few years ago when I redesigned them to have proper input impedance?

ejeffrey, 2015-03-30:
Yes.  I didn't remember when that happened, and it took a while before they
all got replaced.

DanielSank, 2015-03-30:
Ah. Ok, I'm less freaked out now :-)

pomalley, 2015-04-29:
This looks to be the same issue as #97.

closed 2015-04-30 by pomalley