labrad/servers #78: Recovering from timeout doesn't work: ADC's processReadback doesn't have 'executionCounter'

pomalley, 2015-03-28:
See here: https://github.com/martinisgroup/servers/blob/master/GHzDACs/ghz_fpga_server.py#L853 (also, why the unnecessary if clause there?

but here: https://github.com/martinisgroup/servers/blob/master/GHzDACs/Cleanup/adc.py#L1059
(also, why the switch from classmethod in the superclass to staticmethod in the subclass?)

Of course, this would have been much easier to find if the fpga server didn't swallow all its exceptions (see #59).

DanielSank, 2015-03-28:
Woah, wut? Is the execution counter not in the ADC spec? This is Twilight Zone level.

pomalley, 2015-03-28:
Dude, today I went straight from the Twilight Zone of waking up every few hours to feed a newborn to the Twilight Zone of the FPGA server. Now I'm about to open up the ADC documentation...

pomalley, 2015-03-29:
OK, looks like it's just a labeling issue: the ADC counts the number of triggers so the variable is called `trigCount` rather than `executionCounter`.

DanielSank, 2015-03-29:
That is sensible, I suppose.

pomalley, 2015-04-29:
This is fixed in the JT branch, but not in master yet...

closed 2015-04-30 by pomalley