labrad/servers #79: ADC process readback returns numpy bool for noPllLatch

pomalley, 2015-03-29:
This line: https://github.com/martinisgroup/servers/blob/master/GHzDACs/Cleanup/adc.py#L1055

returns a numpy bool. Here's what I get when I print the type:

```
2015-03-28 17:34:37-0700 [ServerProtocol,client] <type 'numpy.bool_'>
```

I don't know at what point this broke.

pomalley, 2015-03-29:
It's easily fixable by casting it to `bool`.

DanielSank, 2015-03-29:
This possibly broke when we made `types.py` less random. Now you get real errors instead of inconsistent unexpected behavior.

That sentence hurt me.

pomalley, 2015-03-29:
Indeed, this was the error message:
```
Error: (0) TypeError("No LabRAD type for: True (type <type 'numpy.bool_'>)",) [p
ayload=None]
```

DanielSank, 2015-03-29:
This is more or less a bug in pylabrad, given that we're officially supporting numpy types. Will fix presently.

DanielSank, 2015-03-29:
Hey, @maffoo, I just went to fix this and I'm a bit puzzled.

DanielSank, 2015-03-29:
@pomalley proposed fix has been submitted in the pylabrad repo.

pomalley, 2015-03-29:
Saw that, thanks.

pomalley, 2015-04-29:
I believe this one is fixed now.

closed 2015-04-30 by pomalley