labrad/servers #80: ADC timeout errors for loop_delay/biasOperateSettling < 5 us

pomalley, 2015-03-30:
In the JT branch of the code, the delay between stats (`biasOperateSettling`) is implemented by way of the loop delay FPGA setting. When this is less than about 5 microseconds, the ADC starts to wack out as described in #77.

I don't know why this is, and I also don't know if this was a problem before--I don't know that we've ever set `biasOperateSettling` this low.

DanielSank, 2015-03-30:
Are you using the new data streaming option in the ADC?

maffoo, 2015-03-30:
I assume you've confirmed from scope traces or whatever that the FPGA boards actually run properly with the shorter loop delay? In the sequences you're running, are the boards sending measurements back with every iteration through the loop? It could be that with such short delays the packets are getting sent too quickly and dropped by the network switch. One way to check this would be to watch the direct ethernet server ui to see the number of packets that are actually buffered at the point where the collect call times out. If it's less than the number you expect even though the boards are looping the correct number of times, then there may very well be ethernet collisions causing packet loss.

pomalley, 2015-03-30:
> Are you using the new data streaming option in the ADC?

I don't know what this means.

@maffoo:

The DACs work fine with the short loop delay; if you set the loop delay to zero they just start the next stat immediately, and the traces look fine on the scope in all cases. Yes, the ADC should be sending back info with every iteration, but we haven't yet tested anything about the ADC w.r.t. this. Ethernet collision is definitely a possibility.

My hunch is that fixing this will take a bit of effort; I made the issue here to see if anyone has ideas for where to look.

maffoo, 2015-04-01:
I think the first thing to look at is how many packets are actually being received by the direct ethernet server. If this number is low, it points to packet collision. Also, what is the rate at which the boards are cycling, e.g. how often would you expect a packet to be coming back? Note that sending a 48-byte timing packet on a 100Mbit ethernet connection takes at least 3.84 microseconds, which is suspiciously close to the ~5 microsecond boundary you're seeing.

pomalley, 2015-04-29:
I think we're going to table this one for now--when we start to need ADC delays of less than 5 us we'll look into it.

OK?

DanielSank, 2015-04-29:
I guess, but this is a bit scary since we don't know why it's happening.

ejeffrey, 2015-04-29:
So I am not super familiar with the way the jump table stuff is
implemented, but the time to send a 50 byte ethernet packet is
approximately 4-5 microseconds.  If you are forcing the ADC to flush the
fifo and send a packet for each stat, you are not going to be able to loop
faster than this.

ejeffrey, 2015-04-29:
Also, I bet that if this is the problem, if you look directly at the output
of the ethernet port with a protocol analyzer you will see it start to send
a packet, then get interrupted in the middle and start with a new packet.
 ​Dan talked to a network engineer from google about various ways to snoop
on a network, including a dedicated hardware tap.  If we had one of those,
this problem could probably be diagnosed immediately.

DanielSank, 2015-04-29:
Indeed, we can get some fancy-pantsy switches if we want. I didn't pull the trigger on that because I think we'd have to learn to use it.

pomalley, 2015-04-29:
Yes, 4 to 5 us is generally the timescale of the FPGA boards responding to ethernet packets so it does not surprise me that this is also the timescale we're seeing here. I'm not sure how exactly the ADC is being told to send back packets but one per stat seems like it's probably what we're doing.

By "table it" I meant not to let it stand in the way of getting the JT PR merged in. (I'd also wager that you'd see this problem in master right now, as well.)

ejeffrey, 2015-04-29:
There may be (probably are) problems on master with short biasOperateSettle
delays, but you would probably never even be able to see this particular
issue as we always add an extra 4.3 microsecond settling delay to the
master board, above and beyond the delay from biasOperateSettle.  The
purpose of this was to delay the master board relative to the slaves so
that it never sent a daisy chain pulse until the slaves were already
waiting for it, but it would have the effect of setting a floor to the time
between stats.