labrad/servers #83: Document GHzDAC calibration procedure.

pomalley, 2015-04-02:
Since @JulianSKelly and I changed the GHzDAC calibration stuff around, there is no longer an interactive script to calibrate the boards. Today Rami wanted to calibrate the boards and I had completely forgotten how we actually do that now. We should document this so that other people know what to do!

DanielSank, 2015-04-19:
+1

I can't figure out how to calibrate the boards on Jules. This needs to happen asap.

DanielSank, 2015-04-19:
@pomalley @JulianSKelly please do this as soon as possible. This is a lab-wide problem.

pomalley, 2015-04-19:
To get you going today:

>>> import servers.ghzdac_cal.GHz_DAC_calibrate as gc
>>> gc.calibrate_iq(cxn, ['Vince DAC 11'])
>>> # or if you want to do all the boards in one go:
>>> gc.calibrate_iq(cxn, gc.find_microwave_dacs(cxn))

DanielSank, 2015-04-19:
Thank you, Peter!

By the way, I note that the option to not use the microwave switch was not used here. I'll add that in.

commit c14b455d (2015-05-18): Add documentation for GHzDAC calibration procedure.

Fixes #83, review: @DanielSank.
PR #158 (2015-05-18): Add documentation for GHzDAC calibration procedure.
Fixes #83.
closed 2015-05-18 by pomalley