labrad/servers #97: Jump table slave mode is broken

JulianSKelly, 2015-04-15:
Using U:/FPGAsof/GHzDAC_V9_B15/GHzDAC_V9_B15_P0_P0_s1.pof

We find that when running the following setup:
Daisy Chain: Vince DAC 19 -disconnected-> Vince DAC 17 -> Vince ADC 1
We set Vince DAC 19 to be master
We set Vince DAC 17 to be slave
We run for 4 stats
Vince DAC 17 claims to have run ~45,000 times, and we see on the scope that the DaisyOut[2] is high for >3 us (should be 4 ns). This happens even though the daisy chain is disconnected between Vince DAC 19 and Vince DAC 17.

This makes us think that slave mode is broken, and sees phantom daisy chain pulses.

DanielSank, 2015-04-15:
1. Run `jtt.test_sine(length=256, num_derps=4, slave=0, stats=1)`. Works. See register write pulse and sine waves on 'scope.
1. `reg = dac.DAC_BUILD13.regRun(reps=1, page=0, slave=1, delay=0, loop_delay=50, sync=0)`. Then `de.write(reg.tostring())`. See register write pulse, and, starting ~3.8us before the rising edge of the register write pulese, the SRAM output starts oscillating and does so indefinitely.

DanielSank, 2015-04-15:
1. `jtt.test_sine(length=256, num_deps=4, slave=0, stats=1)`. See register write pulse followed by sinewave.
1. `reg_master = dac_DAC_BUILD13.regRun(reps=1, page=0, slave=0, delay=0, group_delay=50, sync=0)`, de.write(reg_master.tostring())`. We see another register write pulse, but the SRAM data disappears!

pomalley, 2015-04-16:
The second one could be explained by the `loop_delay`: it seems that the board waits for `loop_delay` us after the register write to fire. So if we were triggering on the register write pulse but didn't have the scope zoomed out far enough, it would disappear. Contrast this to `jtt.test_sine`, where I believe the default loop_delay is 0.

pomalley, 2015-04-16:
I was also able to reproduce the first symptom (slave=1 causing the board to spew junk) on a single board when I had the daisy chain unplugged, that is, both daisy up and daisy down empty. When I put the board in loopback (daisy up into daisy down) then the problem went away, and slave mode did not play the SRAM. This seems to indicate that when the ethernet ports are floating the board is liable to see daisy pulses in slave mode. (I don't know if this is expected, or if ethernet connections are supposed to have some sort of pull-up/pull-down deal.)

Perhaps the master board is not in fact sending pulses on the daisy chain but is only letting it float.

DanielSank, 2015-04-16:
> I was also able to reproduce the first symptom (slave=1 causing the board to spew junk) on a single board when I had the daisy chain unplugged, that is, both daisy up and daisy down empty. When I [...]

Yes! We discussed this possibility at lunch and forget to tell you. Sorry times a million.

> Perhaps the master board is not in fact sending pulses on the daisy chain but is only letting it float.

Yes yes yes. We can figure this out tomorrow.

pomalley, 2015-04-17:
Here's what happens when I wire the test board in loopback mode (run an ethernet cable from daisy_down to it's own daisy_up). The SRAM consists of a sine wave, and the jump table is only one command, END at the end of the sine wave. The register packet is: master start, one rep, mon0 = daisy_in[0], mon1 = daisy_out[2] (and everything else 0).

[image]

Blue is the DAC A+ out, green is mon1 (the daisy out), yellow is mon0 (daisy in), and purple is unused. You can see that the board thinks it is sending a daisy pulse, and then receiving a daisy pulse about 16 ns later, and then playing the SRAM.

pomalley, 2015-04-17:
If I set repetitions > 1, then it repeats this exact sequence the specified number of times.

DanielSank, 2015-04-17:
I just finished a pull request. I'll come help you out now.

pomalley, 2015-04-17:
I did get this to work in a simple test case:

[image]

Purple is daisy out on board 50, and green is daisy in on board 63. They both play their waveforms. This requires further investigation.

pomalley, 2015-04-17:
Also requiring investigation: why github shows the picture sideways.

pomalley, 2015-04-17:
On the rack, I was successfully able to get the daisy chain working on DACs 18->19->17->ADC 1. Also, run_sequence_bloatware ran fine with this as well. Note sure what was failing yesterday.

pomalley, 2015-04-17:
Ran a rabihigh experiment with this using DACs 17, 18, 19 and ADC 1. Things seem to be working.

DanielSank, 2015-04-28:
@pomalley Could you write a quick post-mortem of the problem and then probably close this guy? The problem was that the start bytes have to turn off daily pass-through, right?

pomalley, 2015-04-28:
The problem was that the master board was getting set to slave mode. This was happening with the ping packets being sent to get things like build number, number of executions, etc. The ping packets have now been fixed to put the board in master mode, but with 0 repetitions.

DanielSank, 2015-04-28:
That sounds like a somewhat obscure solution. John could fix this quickly, and it's worth it if it simplifies the code and/or prevents future confusion. He offered to do this at Joshua Tree.

pomalley, 2015-04-28:
I don't think it's particularly obscure--but maybe I didn't explain it well enough. The principle is "don't set a board to slave mode if it has no daisy-in connection". The ping packets were setting all the boards to slave mode. Now they are not. This behavior more closely reproduces the behavior of the old boards, though the specifics of master/slave mode have changed so that it's not a one-to-one correspondence.

We could fix this in hardware, but I think that's more trouble than it's worth, or at least not worth the risk of new bugs.

DanielSank, 2015-04-28:
>  The principle is "don't set a board to slave mode if it has no daisy-in connection". The ping packets were setting all the boards to slave mode. Now they are not.

> The ping packets have now been fixed to put the board in master mode, but with 0 repetitions.

These seem to implicitly contradict each other. I don't wanna be a PITA at 11pm. Can chat for a minute tomorrow.

pomalley, 2015-04-29:
Dan, shall we clear this up, since I'm starting to get the PR ready?

They actually don't contradict each other at all. Here's how it works, as far as my testing has shown:

* Master mode means: start X times, and put out a daisy pulse each time.
* Slave mode means: start when you get a daisy pulse, and pass the daisy through.
* Idle means: pass the daisy pulse through only.
* Open daisy-in connections result in undefined behavior for boards in slave mode.

This means that when we're sending register packets to all the boards (e.g. board discovery at startup, other pings later on, etc), those register packets must put the board in master mode. In fact, this is exactly what was done with the previous version of the boards--the difference is that master mode was defined not by the first byte in the packet, but by one later down, and 0 meant master mode for that byte (whereas now 0 means slave mode, for the first byte).

So the right solution is to set byte[0] = 1 for the miscellaneous (e.g. ping) register packets.

Does this make sense?

DanielSank, 2015-04-29:
Yeah, that makes sense. Thanks.

closed 2015-04-30 by pomalley