labrad/servers #100: ADC Average Mode Broken

jwenner, 2015-04-16:
When try to use fpga average mode, get error in ghz_fpga_server since demod channel not specified. On branch 00d5ad5e77ec5f6dc8030fa0798d063b064bdc4f.
```
In [25]: fpgaTest.average(s,cxn,3,True)
---------------------------------------------------------------------------
Error                                     Traceback (most recent call last)
<ipython-input-25-53c37bbfb41c> in <module>()
----> 1 fpgaTest.average(s,cxn,3,True)

C:\pyle\pyle\scripts\fpgaTest\fpgaTest.py in average(sample, cxn, reps, plot, na
me, dacsAdcs)
   1086     fpga.daisy_chain(daisyChainList)
   1087     fpga.timing_order([adc['_id']])
-> 1088     data = fpga.run_sequence(reps, True)[0]
   1089     if plot:
   1090         plt.figure()

C:\pylabrad\labrad\client.pyc in __call__(self, *args, **kw)
     62         if wrap:
     63             future.addCallback(lambda resp: resp[0][1])
---> 64         return future.wait() if wait else future
     65
     66     # data to be loaded on demand

C:\pylabrad\labrad\backend.pyc in wait(self)
    397                 # If any Future in the queue fails,
    398                 # we immediately bail.
--> 399                 result.raiseException()
    400             else:
    401                 for func, args, kw in f.callbacks:

C:\Python27\lib\site-packages\twisted\python\failure.pyc in raiseException(self)

Error: (0) [GHz FPGAs] Remote Traceback (most recent call last):
  File "C:\pylabrad\labrad\server.py", line 262, in handleRequest
    result = yield setting.handleRequest(self, c.data, data)
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1105, in
_inlineCallbacks
    result = result.throwExceptionIntoGenerator(g)
  File "C:\Python27\lib\site-packages\twisted\python\failure.py", line 389, in t
hrowExceptionIntoGenerator
    return g.throw(self.type, self.value, self.tb)
  File "C:\pylabrad\servers\GHzDACs\ghz_fpga_server.py", line 1449, in run_seque
nce
    timingOrder)
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1107, in
_inlineCallbacks
    result = g.send(result)
  File "C:\pylabrad\servers\GHzDACs\ghz_fpga_server.py", line 758, in run
    raise RuntimeError('channel format not understood')
exceptions.RuntimeError: channel format not understood
 [payload=None]

In [26]:
```

JulianSKelly, 2015-04-16:
There is a hardware bug on the FPGA with respect to average mode, and switching back and forth between average mode and demod mode.

John, myself and others elected not to peruse fixing this bug because we can operate the ADC in 'sampling scope' mode, and couldn't think of an instance when average mode is necessary for any qubit experiments.

DanielSank, 2015-04-16:
@JulianSKelly @ejeffrey @pomalley in that case should the class for this trigger table version of the ADC throw an exception if you try to run it in average mode?

jwenner, 2015-04-17:
FYI, I'll be wanting average mode for my experiment to help with error budgeting (akin to what I've done previously). Otherwise, each time trace will take a ridiculous amount of time when considering the number of stats.

DanielSank, 2015-04-17:
> Otherwise, each time trace will take a ridiculous amount of time when considering the number of stats.

@ejeffrey @JulianSKelly I believe you guys did most of the work with the fpga server support for the trigger table. If you can at least comment on why average mode isn't supported now?

On a broader note: backwards compatibility matters. Future development on the FPGA server should maintain backwards compatibility unless *all* users are involved in a discussion and the group agrees to drop support for something.

JulianSKelly, 2015-04-17:
>There is a hardware bug on the FPGA with respect to average mode, and switching back and forth between average mode and demod mode.

This is why we chose not to support average mode. At that time, John did not want to track down the bug. If the bug has mysteriously disappeared, that is news to me.

DanielSank, 2015-04-17:
Ok, but support was also missing from the FPGA server. Was that on purpose or an accident (this may have been my fault, I just don't remember).

JulianSKelly, 2015-04-17:
I honestly don't remember. Sorry about all of this.

DanielSank, 2015-04-17:
> (Dan) this may have been my fault, I just don't remember

> (Julian) I honestly don't remember. Sorry about all of this.

Heheh, human memory is pretty low fidelity, eh?

ejeffrey, 2015-04-17:
So there is still the problem with the average mode overflowing, which
requires you to rerun the scan a bunch of times to get high stats.

On the other hand, using the demod tables and the readoutSampling scope you
should be able to read out 44 ns chunks at a time (11 demodulators, one
real and imaginary point each = 22 chunks of 2 ns).  With retriggering you
should be able to get the entire waveform in 2 chunks.

DanielSank, 2015-04-17:
Great idea. We should try this and see how fast it is.

commit 5ceae1f5 (2015-04-27): Update return format docstring in ghz_fpga_server.runSequence

Fixes martinisgroup/pyle#103, Fixes #100

Review: @DanielSank @jwenner
commit 670d9069 (2015-04-27): Enable average mode for ADC in ghz_fpga_server.py

In FPGA server's Run and Run Sequence settings, if data
channel is just an ADC board name, e.g. "DR Lab ADC 0",
then we return *3i (board, I/Q, time index) from average
mode.

If a channel is specified as <board name>::channel then
the ADC is run in demodulation mode and data is returned
as a *4i (channel, stat, retrigger, I/Q), just as before.

Bump version number

Fixes #100

Review: @DanielSank
PR #101 (2015-04-27): GHzFPGA.run: Enable ADC Average Mode
Fixes #100.

In FPGA server/run (and hence run_sequence), if ADC in average mode so that no demod channel specified, raised 'channel format not understood' exception. Now add elif case in getTimingData so if ADC is present in channel name but not :: (for demodulation), can still return data. Move calls to channel to if statement for if channel defined. Amend return format type for run_sequence to also allow '*3i' for average mode data.

Note that in going between demod and average modes (although in different scans & calls to run_sequence), it seems to work [... cut]
commit 26cad813 (2015-04-27): Update return format docstring in ghz_fpga_server.runSequence

Fixes martinisgroup/pyle#103, Fixes #100

Review: @DanielSank @jwenner
closed 2015-04-27 by DanielSank