labrad/servers #108: Can we remove GHz_DAC_bringup_needed?

pomalley, 2015-04-20:
It is apparently deprecated.

pomalley, 2015-04-20:
While we're at it let's combine the three ghzdac* folders.

maffoo, 2015-04-20:
before we combine these we need to think about how we want the repo to be organized.

`ghzdac` is a package containing library code for the deconvolution; this code should be importable from elsewhere, so it should have an `__init__.py` and it should be on the python path.

`ghzdac_cal` contains runnable scripts for doing calibration; these scripts should _not_ be in a python package, given python's separation of scripts from library code, and so this should _not_ have an `__init__.py` file. Or at the very least, the scripts should be refactored to put all the logic in a python package, and just have runnable stubs in a separate scripts directory (which is _not_ a python package) that just imports and runs the appropriate library code. The stubs could also do some path manipulation to make sure that the corresponding package directories from the same git checkout are on the python path; this is a concession we've discussed for people who are on windows and want scripts to be double-clickable in a gui environment on a shared machine that may have different people's personal checkouts and so the system python path may point at the wrong location.

Finally, `GHzDACs` contains server code as well as other miscellanea (looks like there are some scripts in there, etc.) Again, anything that should be top-level runnable should be refactored into library code and a script stub in a non-package directory. I think the server itself should also receive the same refactoring treatment, since it needs to be top-level runnable by the node.

So, roughly speaking, I propose the following structure for the servers repo:

```
ghzdac/
    __init__.py
    adc.py
    calibrate.py
    ...

scripts/
    GHz_DAC_bringup.py
    ...

servers/
    ghz_fpga_server.py
    ...
```

Note that we would remove the top-level __init__.py file in the servers repo; this whole thing is not a python package. Also note that the scripts and servers folders do not have `__init__.py` files; these are also not python packages. The servers checkout would no longer have to be put inside an extra directory when checked out and the checkout folder itself could be added to the pythonpath (at least on non-shared machines or in virtualenvs) and this would make the code in the `ghzdac` package available. Also note that the `ghzdac` package as I've sketched it contains both the calibration and server code mashed together; we'd probably want to structure that differently, but I think having a clean separation between scripts and library code (+ tools like pycharm) makes that easier.

DanielSank, 2015-04-20:
One thing at a time. +1 for getting rid of bringup_needed. It was intended as a convenience but in practice it's not worth the bloat it adds to the repo. I vote to remove it.

maffoo, 2015-04-20:
opened a new issue about reorganizing the repo: #109

commit 4f05ccb3 (2015-04-23): Delete GHz_DAC_bringup_needed.py

Fixes #108.

Review: danielsank.
commit 0e12fba2 (2015-04-23): Delete GHz_DAC_bringup_needed.py

Fixes #108.

Review: danielsank.
PR #112 (2015-04-23): Delete GHz_DAC_bringup_needed.py
Fixes #108.
closed 2015-04-23 by pomalley