labrad/servers #114: [Help needed][Jump Table] Frequency error in mixer calibration

JulianSKelly, 2015-04-25:
The jump table update to the boards has somehow broken the (old, fragile, undocumented) iq mixer calibration code. 

The problem is that when we generate a sideband, it appears to be off by up to ~150 KHz on the spectrum analyzer. This is pretty odd, because even if we, for example, didn't account for a 4 ns blip in the SRAM, this shouldn't shift our frequency, right?

This is currently the _only_ thing standing in the way of a jump table pull request, so some brainstorming would be really appreciated.

pomalley, 2015-04-25:
To add to this:

One bit that I'm not sure about is that the data are sent through the deconv function before being sent to the boards. This involves an FFT, and other stuff that I didn't quite follow. It seems unlikely that this is the source of the problem, though, because this part hasn't changed at all.

pomalley, 2015-04-25:
Also I'll add that the way the calibration script works now is to set the spectrum analyzer to a bandwidth of 300 Hz, so previously we seem to have been hitting the frequency pretty closely.

ejeffrey, 2015-04-25:
Is something adding a 4 nanosecond chunk at the beginning or end of the
block?  Adding 4 nanoseconds of 'zero' to a 8 microsecond chunk of a 100
MHz sine wave would shift the frequency down by 50 kHz.

Is the shift up or down?  What does it look like on the scope?

ejeffrey, 2015-04-25:
Also, are there a bunch of close-in sidebands / comb structure on the
spectrum analyzer (say with a 10 Hz RBW)?  Or does it just look like a pure
tone at the wrong frequency?

JulianSKelly, 2015-04-25:
We could easily be off by a 4 ns chuck, as the jump table has some offsets which might not be accounted for in the low level code present in the calibration scripts.

I was under the impression that chuck wouldn't affect the frequency, but I guess I'm mistaken?

I see sidebands off of the sideband, at around 500kHz? Gotta double check.

ejeffrey, 2015-04-25:
If you have 200 cycles over 2 microseconds (for 100 MHz), but instead have
200 cycles over 2.004 microseconds, that will lower your 'fundamental'
frequency.  From the blips every 2 microseconds you would also see
sidebands at +/- 500 kHz, so that sounds like a pretty good candidate for
what is happening.

ejeffrey, 2015-04-25:
[image: Inline image 1]

Here is the FFT spectrum of: 2 microseconds of a 100 MHz sine wave,
followed by 4 samples of zero, repeated.  The fundamental is shifted to
slightly lower frequency and the sidebands are at ~500 kHz intervals.

Evan

JulianSKelly, 2015-04-25:
@ejeffrey you win the signal processing prize, things are working now. Thanks!

@pomalley I had to put the -1 back into dac_run_sram. However, this should already have been accounted for in the jump_table code that we modified. As far as I can tell, we are doing the offsets that the documentation specified. So, we're either doing something subtle or the documentation is wrong.

On monday let's do some careful address budgeting. This should be easy.

pomalley, 2015-04-25:
I'm guessing that there's another offset we need to figure out in faux-loop
mode  (i.e. Jump back to zero). I'll think about it.

pomalley, 2015-04-29:
I believe we can close this.

closed 2015-04-29 by JulianSKelly