labrad/servers #115: Tests for the GHz FPGA server

pomalley, 2015-04-27:
Commit 3c880369f64efa6616476ee47b1a5035fb454ec7 has tests for the jump table byte generation, which are pretty basic, but I'm pretty excited about commit 45f1d41e0095951a9639ef9d8328b300684d07b3 which has tests for the FPGA server. It's not super-pretty at the moment because it involves creating a server object without a Twisted reactor, but I think it gets the job done for testing the logic of many of the FPGA server calls. It uses the mock library to read the calls to the direct ethernet server, and then tests the packets for correctness. In fact, the tests currently fail in this branch because the FPGA hardware offsets for the jump table are getting subtracted twice, once in the low level jumpTable.py and once at a much higher level in the FPGA server settings. So the tests have caught their first bug!

I'd like others to take a look at it to see if this makes sense. The tests are incomplete at the moment and could certainly stand some refactoring. (In fact, if anyone wants to add some tests to this that would be great, because my schedule is about to get a lot more cramped.)

Now I'm going to tag a bunch of people because I'm excited about this: @JulianSKelly @DanielSank @ejeffrey @maffoo.

DanielSank, 2015-04-27:
Yayayayayayay!

pomalley, 2015-04-28:
I cleaned up the tests a bit and also fixed the tests by removing the repeated offset correction. I think I'm done with the tests for the moment.

pomalley, 2015-05-16:
> I think I'm done with the tests for the moment.

closed 2015-05-16 by pomalley