labrad/servers #119: PR incoming for the jump table stuff

pomalley, 2015-04-29:
As we discussed elsewhere, I'm going to be breaking up the u/pomalley/jump-table2 branch into a series of smaller diffs, in preparation for code review. If you've got anything to say about any of the files touched by this branch, speak now or forever hold your peace! (Does anyone ever actually "speak now", outside of movies?)

The first step is (probably) going to be rebasing this off master and force-pushing, so this will likely invalidate anyone's cloned version of this branch. Sorry.

pomalley, 2015-04-29:
The PR's have been created. See #122 and #123.

DanielSank, 2015-04-29:
:star: Yayayayay :star:

jwenner, 2015-04-29:
Settings 1300 and 1301 (DAC Bringup and DAC Bringup New) have the same function name.

jwenner, 2015-04-29:
Ignore the most recent comment. Meant to put it on the PR, not the issue.

pomalley, 2015-05-16:
This has been merged. Thanks everyone for the reviews!

closed 2015-05-16 by pomalley