labrad/servers #127: What words should we use for the various sized chunks of SRAM data?

DanielSank, 2015-05-04:
The elements we need to name are:

1. 32 bit chunks representing one DAC cycle's worth of data. These have been called "words" up to the present time, and I think this is a good word to use. In digital circuits, "word" is commonly used as the smallest size chunk larger than a single bit. I think there will not be any disagreement, so I'll call this a "word" for the rest of the post.

1. Consecutive groups of four 32-bit words. This is the smallest size chunk the FPGA sends to the DAC. These four word chunks form the smallest element used with in-fpga addressing, i.e. in the jump table's start_addr, etc.

1. Groups of 256 words. This is the chunk of data written in one ethernet packet to the board. This used to be called a "page" until I moronically changed it to "derp" to avoid conflicting with the other (now outdated) use of the word "page" in the fpga server.

I propose, "word", "phrase", and "page".

I previously have called groups of four words a "cell", and I actually think that's a more descriptive word than "phrase", but parallel structure is important.

jwenner, 2015-05-04:
Or if you don't want to reuse "page", perhaps "sentence" or "paragraph"?

maffoo, 2015-05-04:
+1 for "word", "cell" and "page". I feel like "phrase" stretches the linguistic metaphor a bit too far.

ejeffrey, 2015-05-04:
Agree with Matthew

DanielSank, 2015-05-05:
Ok, glad we all agree (I like cell more but didn't propose it up front because of the lack of parallelism).

Let' make it official:

* **word:** 32 bits of SRAM data representing the DAC output for one clock cycle (1 ns).

* **cell:** A contiguous group of 4 words. The starting word of a cell always has index `4*n` for some integer `n`. A cell represents the finest amount by which fpga features like the jump table can index the SRAM.

* **page:** 256 SRAM words. The starting word of a page always has index `256*n` for some integer `n`. A page is the unit by which data is written to the board (via ethernet).

When possible, let's all convert the documentation and variable names to conform to this.

DanielSank, 2015-05-09:
Closing, since we've agreed on terminology.

closed 2015-05-09 by DanielSank