labrad/servers #134: ghz_fpga_server fails to reconnect when direct ethernet server is restarted

maffoo, 2015-05-08:
We don't normally do this, but since we've been trying out the scala version of the direct ethernet server we've been restarting and noticed that after connecting to a board group once, when the direct ethernet server is restarted the ghz_fpga_server does not properly reconnect, so the devices don't appear again. Restarting both servers works, but is annoying.

strj, 2015-06-01:
I may have completely misread the code but looking into this it seems that the issue is that when the direct_ethernet (DE) server reconnects, the currently existing board groups in the ghz_fpga_server are not updated with the new DE server. Matthew, does that sound about right to you?

maffoo, 2015-06-01:
The DE server object itself should not need to get updated. That is just a proxy and since it reuses the network connection to the manager, it will happily talk to the new direct ethernet server after it restarts. However, the board groups will need to reinitialize their connection to the server, for example setting up their contexts with the appropriate packet filters.

maffoo, 2015-06-18:
@zchen088 saw another case where the ghz_fpga server did not reinitialize properly after the direct ethernet restarted. Got an error back when trying to take data after an ethernet server restart:
```
Traceback (failure with no frames): <class 'labrad.types.Error'>: (0) java.lang.RuntimeException: context: Context(18,14). must connect to an adapter first
       at scala.sys.package$.error(package.scala:27)
```
So the ghz_fpga server was trying to talk to the ethernet server but hadn't connected to an adapter.

strj, 2015-06-18:
I introduced some typos with #179 that I fixed only a couple of hours ago - is this with the latest ghz_fpga_server?

maffoo, 2015-06-18:
The code is at 4cd4b3cd, so does not include the style cleanups from #179.

strj, 2015-06-25:
I've been looking into this and I can't work out why https://github.com/martinisgroup/servers/blob/36028cead210791257c8abe49147118843a4e3f5/GHzDACs/ghz_fpga_server.py#L290
wouldn't reconnect the boards.

Unless the error happens due to an asynchronous call before the connect call is made. It seems unlikely and I'm not sure it's possible?
Or there's something wrong upstream... is there more output before the error?

I'm OOO for the next two weeks but can look into this more when I get back

commit a1fce054 (2015-06-04): Reinitialize the all the BoardGroup's on Direct Ethernet restart.

Move the BoardGroup Init call to reinitialize existing BoardGroups on DE server
reconnecting.

Fixes #134
PR #174 (2015-06-04): Reinitialize BoardGroup's on Direct Ethernet restart.
Switches the BoardGroups to get context and connection from the
DE server every time before using either. Move the BoardGroup
Init call to aslo reinitialize existing BoardGroups on
DE server reconnecting.

Fixes #134
commit 172b3372 (2015-10-06): Wait for packet capture thread to start before returning from listen.

Fixes #232 and fixes #134
Review: @jwenner
closed 2015-10-06 by maffoo