labrad/servers #135: fpga server timeout recovery raises KeyError on 'executionCounter'

DanielSank, 2015-05-09:
This is happening because the ADC class doesn't set `executionCounter` in the readback data.
I'm gunna just make it gracefully handle the case where `executionCounter` is not available.

Future work should implement the execution counter on any boards which support it!
I think this isn't implemented properly now out of laziness (probably laziness of past Dan).

jwenner, 2015-05-09:
1. I believe this is equivalent (but stated better than) #99.
   1. @pomalley, I believe you had a fix that you said was implemented in the jump table branch. What was it?

DanielSank, 2015-05-09:
I already fixed this. PR incoming.

pomalley, 2015-05-09:
Question: I had "fixed" this in the JT branch by changing the ADC process readback function for build 7--specifically, I renamed the key `trigCount` to `executionCounter` because I thought it was a typo/bug. Is it in fact the case that the ADC build 7 reports the number of triggers, not executions, and thus my fix was incorrect?

DanielSank, 2015-05-09:
I think you did the right thing, Peter. Can we close?

pomalley, 2015-05-09:
Sure.

commit 5b14530a (2015-05-09): Gracefully handle unavailable executionCount

Fixes #135
commit 53395744 (2015-05-09): Gracefully handle unavailable executionCount

Fixes #135

Review: ChristopherQuintana
PR #136 (2015-05-09): Gracefully handle unavailable executionCount
The server's recover from timeout strategy involves reading an execution counter form the boards. Unfortunately, not all boards support this and the server was failing on a `KeyError`.
Now we gracefully handle this case.

Tested running on Jules, got timeout, server reported execution counts and carried on!

Fixes #135.
closed 2015-05-09 by pomalley