labrad/servers #142: Why does mem sync delay have to be called last in qubit sequencer memory sequence?

DanielSank, 2015-05-12:
The doc string says "Call last in the memory sequence". Is this a suggestion or a requirement for some reason?

maffoo, 2015-05-12:
This is not a requirement, just the typical use case. What the call does is add a delay (or no-op) to make all the memory sequences the same length, in case they have different lengths so far. Depending on what you are doing, it might make sense to synchronize everything and then continue adding more memory commands. We could change the doc string to clarify.

closed 2015-05-12 by DanielSank

ejeffrey, 2015-05-12:
This is closed, but I want to add a comment.

For reliable operation, mem_sync_delay should be called before run_sram,
and also at the end of the sequence.  It _may_ also be called anywhere else
as needed, but those two calls need to be there to make sure the master
doesn't send the start pulse before the slave boards are waiting at the run
sram opcode.