labrad/servers #145: Are there instructions on how to write a server?

joshmutus, 2015-05-12:
So I'm trying to write a new server, but besides just "copying an existing server" I don't know where to start. Is there documentation for this?

Obviously,
if __name__ == '__main__':
    from labrad import util
    util.runServer(__server__)

is important, but beyond this, I'm kind of lost.

ejeffrey, 2015-05-12:
That is a good point.  I will write up a brief readme on this if you will
review it.

DanielSank, 2015-05-12:
@ejeffrey what's wrong with [this](https://github.com/martinisgroup/pylabrad/wiki/Writing-Servers)?

joshmutus, 2015-05-12:
Oh, that's great. Should this be in servers?

ejeffrey, 2015-05-12:
Oh right.  I forgot about that.  Maybe we can put a stub in the servers
repo that has a link to that page?

DanielSank, 2015-05-12:
> Maybe we can put a stub in the servers repo that has a link to that page?

+1

DanielSank, 2015-05-16:
@joshmutus 

> Oh, that's great. Should this be in servers?

That would make sense. Github's auto-generated wiki is a git repo apart from the actual source. I agree that this is a bit weird, but I'm not entirely sure how to fix it. I guess we could just copypasta and remember not to use the web interface for the wiki (or maybe we can redirect it to a sub-directory of the source repo!).