labrad/servers #151: CYCLE operation bug?

bmyersq, 2015-05-14:
We ran into what seems to be a bug.

Our waveform is

waveform = np.zeros(16384)
waveform[0:2000] = 1

and is then converted to bytes for the sram.

For the jump table, we have

fpga.jump_table_add_entry('NOP', 16)
fpga.jump_table_add_entry('CYCLE', (5000, 0, 1, 0))
fpga.jump_table_add_entry('END', 6000)
fpga.jump_table_set_counter([5, 0, 0, 0])

the result is shown on first figure.

[image]

If we set

fpga.jump_table_add_entry('NOP', 80)
fpga.jump_table_add_entry('CYCLE', (5000, 0, 1, 0))
fpga.jump_table_add_entry('END', 6000)
fpga.jump_table_set_counter([5, 0, 0, 0])

the result is shown on second figure.

[image]

Basically, it seems that if the fromAddress of the 'NOP' operation is too small, the 'CYCLE' operation doesn't work as expected.

pomalley, 2015-05-14:
Yes--there is a hardware limitation that from addresses must be spaced at least 2 cells from each other. The from address of the NOP operation in the first example is 16 ns, which means cell 4. The convention is to subtract 2 from all from addresses, so the 4 becomes a 2 which is too close to the start address of 0. So we get undefined behavior, which you see here.

So this isn't a bug with cycle, but rather a hardware limitation.

How can we better document this?

bmyersq, 2015-05-14:
We use a lot the file JumpTableIntro.md to understand how to run the board, it might be useful to document this in that file. Maybe also add a comment on the jump_table_add_entry documentation? Or raise an error when fromAddresses get too close?

Thanks for help.

maffoo, 2015-05-14:
The server should raise an error if you try to run something like this, that is, it should know about and enforce these hardware constraints. I agree documentation is good, but we shouldn't rely on users reading the docs and only submitting valid commands. @pomalley, would it be difficult to make this happen?

DanielSank, 2015-05-14:
+1 for code that helps you.

pomalley, 2015-05-14:
OK, see the referenced commit and the previous one for code and documentation fixes.

DanielSank, 2015-05-16:
@pomalley I'm pretty sure we can close this.

pomalley, 2015-05-16:
Sure thing.

commit 914e0ecc (2015-05-14): Check from_addr spacing in JT commands.

Fixes #151.
closed 2015-05-16 by pomalley