labrad/servers #156: dc Rack server has several misuses of @inlineCallbacks and ReturnValue

DanielSank, 2015-05-15:


zchen088, 2015-05-16:
Can you elaborate on what that means? I don't know what either of those things is. Also what are the offending functions?

DanielSank, 2015-05-16:
> Also what are the offending functions?

* `returnCardList`

* `preampState`

* `getMonitorState`

* `initServer` had a misplaced `yield`

* `preampChannelState` had a bad `returnValue`

> Can you elaborate on what that means? I don't know what either of those things is.

This is a bit of a big question for a github issue because it has to do with how Twisted (our asynchronous even loop library) works. The basic gist of it is that you want to make a network call in your server, but you don't want the server to stop serving other requests while waiting for the response to that network call. There are two general strategies to achieve this: threads and cooperative multitasking.

### Threading

In threading the idea is that the thread scheduler can decide when to execute code paths in different threads. It's nice because you don't have to decide when to give up control in your code, but it's not nice because the scheduler can switch you at any time, so operating on any memory shared between threads is turbo tricky. Typically you use "thread safe" objects like locks queues to pass data between threads. We don't do this in pylabrad for several reasons.

### Cooperative multitasking

We use cooperative multitasking. In cooperative multitasking the idea is that your code decides when to give up control, and control is given *back* to your code by the event loop whenever the data you need to keep going is available *and whoever else was running gives up control*.

Example: I'm running a computation, but now I need to know the value of something in the registry. I ask the event loop to make the network call and wake me up when the data comes in. Meanwhile, the event loop goes and executes someone else. If that person never gives up control, I never get my data. That's a bug. If the other guy makes a network request and gives up control, the even loop wakes up whoever's data comes in first and that part of the code executes until completion or until it yields control again.

How does this actually work?

The short version is that you decorate your function with `@inlineCallbacks`, put `yield` in front of every network call, and use `returnValue` instead of `return`. Doing this ensures that the event loop handles network communication in the right way and injects the response data back into the function appropriately.

[More information here](https://github.com/martinisgroup/pylabrad/wiki/Asynchronous-programming-HOWTO).

closed 2015-05-18 by DanielSank

amopremcak, 2016-02-08:
When you say a network call, does that include any call that uses a setting from another server, regardless of whether or not this particular setting uses ```return``` or ```returnValue```? I'm going to read asynchronous programming intro you recommended above but I am looking for a quick rule of thumb to tell these undergrads that are currently helping me write servers.

maffoo, 2016-02-08:
@amopremcak, "network call" really means any time you call something which returns a twisted Deferred. In practice, this means any time you call a setting on another server. It doesn't matter whether that code uses `return` or `returnValue` internally, when you call it over the network with labrad with an asynchronous fashion (for example, when you're writing a server) labrad will give it to you as a Deferred, you have to `yield` inside `inlineCallbacks` to get the value that Deferred produces.

amopremcak, 2016-02-08:
Thanks @maffoo.

commit bdaacb00 (2015-05-18): Remove misuses of @inlineCallbacks and convert associated returnValue to return

Fixes #156

Review: @pomalley
PR #157 (2015-05-18): Fix synchronous versus asynchronous calls in dc Rack Server
Some functions were decorated by `@inlineCallbacks` for no reason, so we remove that and convert associated `returnValue` calls to `return`.

Fixes #156.

(Might fix #139)