labrad/servers #161: Memory usage of direct ethernet, qubit server

rb25, 2015-05-20:
At Vince, we've been running into some slow datataking issues, related to pushing the memory of redheat.

Below are screenshots from the processes for three cases:
-running RB, after running RB for over a the whole night
   mem use direct ethernet: 315 MB
   mem use qubitserver: 1 GB
   mem use labrad: 743 MB
-stopped all scans (no measuring going on for a few minutes), didn't restart any server
   mem use direct ethernet: 435 MB
   mem use qubitserver: 535 MB
   mem use labrad: 681 MB
-restarted servers: direct ethernet, qubit sequencer and fpgaserver, and running RB
   mem use direct ethernet: 184 MB
   mem use qubitserver: 549 MB
   mem use labrad: 642 MB

There might be two things going on at the same time:
- the qubitserver doesn't have a memory leak, but it's garbage collection only acts when it is doing nothing for a long time, and the memory use keeps growing
- the direct ethernet server may have a memory leak

Below are the screenshots:
[image]

[image]

[image]

And the memory use during RB pushing the limits:

[image]

maffoo, 2015-05-20:
@rb25 can I get a block of time on the fridge to do some heap profiling on the servers? I'd like to see if we can figure out whether there really is a memory leak and/or how we should tune the garbage collector.

rb25, 2015-05-20:
Totally. If you want the whole of today, I'm fine with that. Let me know if you need any help.

maffoo, 2015-05-20:
I did some heap and cpu profiling on the server processes on redheat and didn't see any evidence of a memory leak. I did find that the default maximum heap size is 1G on 64-bit windows with our version of java, so that's why the processes were climbing up to about that level in some cases. I've set the max heap size for the server down to 600M which should prevent java from eating up all 4 gigs on the box and slowing things down. We can monitor and adjust as needed.

closed 2015-05-20 by maffoo