labrad/servers #164: Convert Qubit Server to scala

maffoo, 2015-05-21:
We'd like to convert the qubit server to scala, to consolidate on a single JVM language and to take advantage of the performance improvements in the scalabrad api. Because this is a big change, we'll be taking it in stages, and will use this issue to track the overall progress. The plan is roughly as follows:

1. Translate chunks of files from java to scala. Doing this in chunks will keep the size of reviews small, and also allow us to maintain a working system at each stage. Because we'll be interoperating with java and scala, we won't be able to use a lot of the scala goodies, but we'll keep things very literal here.
2. Once everything is in scala, we can add in some of the scala goodies to simplify the code, use scala collections, etc. Again, we'll do this in small chunks so that reviews are manageable.
3. Convert from jlabrad to scalabrad.

jwenner, 2015-11-26:
@maffoo, with #250 in, does this mean that this issue can be closed?

maffoo, 2015-11-30:
Yup, thanks @jwenner!

closed 2015-11-30 by maffoo