labrad/servers #177: Hierarchy of server restarts

rb25, 2015-06-04:
I want to connect the fast scope.
When I connect a new device, I need to restart the GPIB bus server - I do that.

The scope server doesn't see it yet. So I restart the GPIB Device manager

Now the scope server sees it. Great.

But now the Lakeshore_ruox, Lakeshore_diodes, and Anritsu server need to be restarted. Not the hittite, that one is ok.

I restart the above 3 servers.

Now Supermonitor crashes.

Is this how it's supposed to work?

maffoo, 2015-06-04:
You shouldn't need to do any restarts. You should be able to just tell the bus server to rescan, and then all the device servers will detect this and you're on your way:
```python
>>> cxn.vince_gpib_bus.refresh_devices()
>>> cxn.scope_server.list_devices()
...
```
In the past, the gpib server would periodically scan for new devices, but I think folks have disabled that because the periodic scanning can be disruptive, so you just have to trigger it manually. If that doesn't work, then we may need to fix the autodetect logic in some servers, but it shouldn't be too hard.

As for the supermonitor dying when servers restart, that is something we should fix, and worth filing a separate issue about.

rb25, 2015-06-04:
Thanks! Didn't know this got fixed

rb25, 2015-06-04:
But separately: if you restart GPIB device manager, do we expect some servers to fail and others not? I think that's not right

maffoo, 2015-06-04:
It definitely is not right. If things don't properly reconnect after a server restart, we should track down why and fix those issues.