labrad/servers #182: Scala direct ethernet server crashes.

pomalley, 2015-06-08:
We're using version 1.0.0-M1 on redheatdos and it crashes periodically. There is no trouble starting it again with the node but it's rather annoying. 

... @maffoo?

maffoo, 2015-06-08:
A couple of things to try:
* Run the server from the command line instead of the node. Is there any output when the server process exits?
* Try running the 1.0.0-beta package which was running previously on redheat (it should be available in S:\labrad). Do you see the same behavior with that package? If that package is ok, then it could be a problem with the build, and we'll have to figure that out.
* Try running the delphi direct ethernet server. Do you see the same behavior there? If the delphi has problems too, then it could be a problem with the new hardware, perhaps a driver issue with one of the network cards.

JulianSKelly, 2015-06-08:
```
C:\labrad\servers\DirectEthernet-1.0.0-M1>java -cp lib/* org.labrad.ethernet.ser
ver.DirectEthernet
13:27:35.518 [main] WARN  org.pcap4j.util.PropertiesLoader - [org/pcap4j/pcap4j.
properties] Could not get value by org.pcap4j.af.inet, use default value: null
13:27:35.528 [main] WARN  org.pcap4j.util.PropertiesLoader - [org/pcap4j/pcap4j.
properties] Could not get value by org.pcap4j.af.inet6, use default value: null
13:27:35.528 [main] WARN  org.pcap4j.util.PropertiesLoader - [org/pcap4j/pcap4j.
properties] Could not get value by org.pcap4j.af.packet, use default value: null

13:27:35.528 [main] WARN  org.pcap4j.util.PropertiesLoader - [org/pcap4j/pcap4j.
properties] Could not get value by org.pcap4j.af.link, use default value: null
13:27:35.528 [main] INFO  org.pcap4j.core.Pcaps - 4 NIF(s) found.
13:27:35.538 [main] WARN  org.pcap4j.util.PropertiesLoader - [org/pcap4j/pcap4j.
properties] Could not get value by org.pcap4j.dlt.raw, use default value: null
13:27:35.548 [main] INFO  org.labrad.ServerConnection$ - LABRADNODE -> Vince
13:27:35.848 [main] INFO  org.labrad.ServerConnection - Now serving...
13:27:36.068 [Thread-0] INFO  org.pcap4j.core.PcapHandle - Starting loop.
13:27:36.078 [Thread-0] WARN  o.p.packet.factory.PacketFactories - java.lang.NoC
lassDefFoundError:org/pcap4j/packet/factory/PacketFactoryBinder
org.pcap4j.core.PcapNativeException: Error occured: read error: PacketReceivePac
ket failed
        at org.pcap4j.core.PcapHandle.doLoop(PcapHandle.java:809)
        at org.pcap4j.core.PcapHandle.loop(PcapHandle.java:778)
        at org.pcap4j.core.PcapHandle.loop(PcapHandle.java:743)
        at org.labrad.ethernet.server.EthernetBus$$anon$1.run(EthernetBus.scala:
62)
Exception in thread "Thread-0" org.pcap4j.core.PcapNativeException: Error occure
d: read error: PacketReceivePacket failed
        at org.pcap4j.core.PcapHandle.doLoop(PcapHandle.java:809)
        at org.pcap4j.core.PcapHandle.loop(PcapHandle.java:778)
        at org.pcap4j.core.PcapHandle.loop(PcapHandle.java:743)
        at org.labrad.ethernet.server.EthernetBus$$anon$1.run(EthernetBus.scala:
62)
```

maffoo, 2015-06-08:
In looking at the winpcap docs, they say this kind of error can be caused by things like non-standard firewall configurations: http://www.winpcap.org/misc/faq.htm#Q-10

Are the firewall and NIC settings the same on redheatdos as they were on redheat?

pomalley, 2015-06-08:
I don't know. I disabled the firewall on the gigabit adapter but not the others--perhaps we should also disable it on the DAC and GPIB adapters as well.

maffoo, 2015-06-09:
It should definitely be disabled on the DAC adapter; probably fine to leave it on the GPIB adapter. I'm creating a patch that will keep the server running in case an error like this occurs; it may cause packet loss, but things will just time out and there will be no need to restart any servers.

pomalley, 2015-06-09:
I just now disabled it on the DAC adapter. (Turns out I was wrong, it wasn't disabled on the gigabit one, so the DAC one is the only one it's disabled for.)

JulianSKelly, 2015-06-22:
```
00:04:42.802 [EthernetBus-listenThread-\Device\NPF_{2C339376-DD48-493F-A338-5472
5A07704D}] ERROR o.labrad.ethernet.server.EthernetBus - error in listen thread f
or bus \Device\NPF_{2C339376-DD48-493F-A338-54725A07704D}
org.pcap4j.core.PcapNativeException: Error occured: read error: PacketReceivePac
ket failed
        at org.pcap4j.core.PcapHandle.doLoop(PcapHandle.java:813) ~[pcap4j-core-
1.5.0.jar:na]
        at org.pcap4j.core.PcapHandle.loop(PcapHandle.java:782) ~[pcap4j-core-1.
5.0.jar:na]
        at org.pcap4j.core.PcapHandle.loop(PcapHandle.java:747) ~[pcap4j-core-1.
5.0.jar:na]
        at org.labrad.ethernet.server.EthernetBus$$anon$1.run(EthernetBus.scala:
65) ~[directethernet_2.11-1.1.0.jar:1.1.0]
00:04:42.802 [EthernetBus-listenThread-\Device\NPF_{2C339376-DD48-493F-A338-5472
5A07704D}] INFO  org.pcap4j.core.PcapHandle - Closed.
Exception in thread "EthernetBus-listenThread-\Device\NPF_{2C339376-DD48-493F-A3
38-54725A07704D}" org.pcap4j.core.PcapNativeException: Return code: -1, Message:
 PacketGetAdapterNames: The data area passed to a system call is too small. (122
)
        at org.pcap4j.core.Pcaps.findAllDevs(Pcaps.java:68)
        at org.pcap4j.core.Pcaps.getDevByName(Pcaps.java:135)
        at org.labrad.ethernet.server.EthernetBus.org$labrad$ethernet$server$Eth
ernetBus$$reopen(EthernetBus.scala:110)
        at org.labrad.ethernet.server.EthernetBus$$anon$1.run(EthernetBus.scala:
63)
```

JulianSKelly, 2015-07-01:
This morning windows had disabled the Ethernet adaptor for the FPGAs. Any chance this could be a weird side effect of an updated version of winpcap? This behavior seems really strange.

maffoo, 2015-07-06:
Just heard from @JulianSKelly that this seems to have been a result of power management settings causing the network adapters to be shut down by windows. Can this issue be closed?

zchen088, 2015-07-07:
Yes I think it can be closed.

I think the moral of the story here is to use Windows Event Viewer when you suspect the OS of doing funny things.

commit 2ea0276e (2015-06-09): Keep pcap loop alive even if errors occur.

Fixes #182
PR #186 (2015-07-07): Keep pcap loop alive even if errors occur.
Fixes #182 

I think this is probably worth having even if the firewall configuration changes fix the pcap errors. We will want to monitor the logs for any errors that crop up in case this happens frequently, but that's a whole other thing...
closed 2015-07-07 by maffoo