labrad/servers #183: Supermonitor should recover when servers restart

maffoo, 2015-06-08:
It's annoying to have to restart the supermonitor because some servers were restarted. It should just reestablish whatever connections it needs and continue working.

pomalley, 2015-06-08:
It's supposed to do that, but it doesn't always work. The problem is that there are three pieces of code and the responsibilities are not particularly intelligently divided between SuperMonitor.py, LabRADPlotWidget3.py, and the dr_logger itself.

pomalley, 2015-06-09:
FWIW the lakeshore ruox server died on redheatdos (I have no idea why, just got a "python.exe has stopped working" popup) and when I restarted it the DR logger did fine. I don't know how the supermonitor did.