labrad/servers #184: ADR Server Won't Start

BrooksCampbell, 2015-06-08:
Trying to run ADR.py server in console or from web interface.  Servers and Pylabrad are current with master.

Here is the traceback from the console.

C:\labrad-servers\servers\ADR>python adr.py
2015-06-08 14:03:53-0700 [-] Log opened.
2015-06-08 14:03:53-0700 [-] Starting factory <__main__.ADRServer instance at 0x
05765580>
2015-06-08 14:03:53-0700 [Uninitialized] connected to 127.0.0.1:7682
2015-06-08 14:03:53-0700 [ServerProtocol,client] ADR Server starting...
2015-06-08 14:03:53-0700 [ServerProtocol,client] device quaid missing nodes: ['g
ordy']
2015-06-08 14:03:53-0700 [ServerProtocol,client] device rottmayer missing nodes:
 ['collateral']
2015-06-08 14:03:53-0700 [ServerProtocol,client] device victor missing nodes: ['
collateral']
2015-06-08 14:03:53-0700 [ServerProtocol,client] device backups missing nodes: [
'a']
2015-06-08 14:03:53-0700 [ServerProtocol,client] new_status: cooling down, _stat
us:
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: A
DR hauser the second status is now: cooling down
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et PIDloopCount to 0
2015-06-08 14:03:53-0700 [ServerProtocol,client] {'lakeshore': ('lakeshore_diode
s', 'terminator'), 'temperature_lockin': ('SRS lockin', 'terminator'), 'magnet':
 ('agilent_6641a_ps', 'terminator'), 'compressor': ('cp2800_compressor', 'termin
ator'), 'lakeshore_diodes': ('lakeshore_diodes', 'terminator'), 'heatswitch': ('
heat_switch', 'terminator'), 'PNA': ('agilent_n5230a_pna', 'terminator')}
2015-06-08 14:03:53-0700 [ServerProtocol,client] connected to terminator GPIB Bu
s - GPIB0::12::INSTR for lakeshore
2015-06-08 14:03:53-0700 [ServerProtocol,client] Server SRS lockin does not have
 device terminator
2015-06-08 14:03:53-0700 [ServerProtocol,client] connected to terminator GPIB Bu
s - GPIB0::5::INSTR for magnet
2015-06-08 14:03:53-0700 [ServerProtocol,client] Server cp2800_compressor does n
ot have device terminator
2015-06-08 14:03:53-0700 [ServerProtocol,client] connected to terminator GPIB Bu
s - GPIB0::12::INSTR for lakeshore_diodes
2015-06-08 14:03:53-0700 [ServerProtocol,client] Server heat_switch does not exi
st. Check that the server is running and refresh this ADR
2015-06-08 14:03:53-0700 [ServerProtocol,client] Server agilent_n5230a_pna does
not exist. Check that the server is running and refresh this ADR
2015-06-08 14:03:53-0700 [ServerProtocol,client] Loading state variables from su
bdirectory: ruox measurement
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: L
oading state variables from subdirectory: ruox measurement
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et lockinOhmsPerVolt to 100.0 kOhm/V
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et resistanceCutoff to 1725.78 Ohm
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et ruoxChannel to 4
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et lowTempRuoxCurve to 1 / (p[0] + p[1] * r * np.log(r) + p[2] * r**2 * np.log(r
))
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et ruoxCoefsHigh to (-0.3199412, 5.74884e-08, -8.8409e-11)
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et ruoxCoefsLow to (-0.77127, 0.00010068, -1.072e-09)
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et interpolationData to (['', 'ADR', 'Ruox Calibration'], 8L)
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et highTempRuoxCurve to 1 / (p[0] + p[1] * r**2 * np.log(r) + p[2] * r**3)
2015-06-08 14:03:53-0700 [ServerProtocol,client] Loading state variables from su
bdirectory: state variables
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: L
oading state variables from subdirectory: state variables
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et PIDsetTemp to 0.1 K
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et voltageStepDown to 0.006 V
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et maxCurrent to 9.0 A
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et voltageStepUp to 0.006 V
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et targetCurrent to 8.0 A
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et PIDstepTimeout to 5.0 s
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et tempRecordDelay to 10.0 s
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et PIDcp to 1.0 V/K
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et rampWaitTime to 0.2 s
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et recordTemp to False
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et cooldownLimit to 3.9 K
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et quenchLimit to 4.0 K
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et PIDcd to 25.0 V/K
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et PIDintLimit to 0.4 K
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et autoRecord to True
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et loglimit to 20
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et PIDci to 0.4 V/K
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et voltageLimit to 0.28 V
2015-06-08 14:03:53-0700 [ServerProtocol,client] WARNING: not setting previously
 undefined state variable switchPosition to 2
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et autoControl to False
2015-06-08 14:03:53-0700 [ServerProtocol,client] WARNING: not setting previously
 undefined state variable lockinCurrent to 1.0 A
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et fieldWaitTime to 45.0 min
2015-06-08 14:03:53-0700 [ServerProtocol,client] stardate 2015-06-08 14:03:53: S
et recordingTemp to 250.0 K
2015-06-08 14:03:54-0700 [ServerProtocol,client] There was an error: cannot impo
rt name _fblas
2015-06-08 14:03:54-0700 [ServerProtocol,client] Stopping factory <__main__.ADRS
erver instance at 0x05765580>
2015-06-08 14:03:54-0700 [-] Main loop terminated.

pomalley, 2015-06-08:
"cannot import name fblas" makes me think numpy or scipy isn't installed?
or not correctly installed?

BrooksCampbell, 2015-06-08:
Before I started taking data on the fridge a couple weeks ago, I had to upgrade to python 2.7.9 and I uninstalled all the python 2.6/2.7 and packages and installed everything from the dataTakingInstall here:  S:\DataTakingInstall

I'm not sure what I changed from the last cooldown when it was working.

```$ import scipy

$ scipy.version.version
'0.14.0'```

![image](https://cloud.githubusercontent.com/assets/10713905/8045494/89beae22-0de8-11e5-8722-506852431e4d.png)

pomalley, 2015-06-08:
I don't know why it's not working now, but we can try doing what they say here:

http://stackoverflow.com/questions/25406492/importing-scipy-or-scikit-image-from-scipy-linalg-import-fblas-import-error

BrooksCampbell, 2015-06-08:
I installed 0.14.1 (http://sourceforge.net/projects/scipy/files/scipy/0.14.1/) perhaps someone more knowledgeable can pick the correct version for us--0.16 is in beta and 0.15 is available.

closed 2015-06-08 by BrooksCampbell