labrad/servers #194: FPGA server timeout recovery broken

pomalley, 2015-06-25:
At least I think that's the problem:

```
Error: (0) java.util.concurrent.ExecutionException: org.labrad.errors.LabradException: [GHz FPGAs] Remote Traceback (most recent call last):
  File "C:\pylabrad\labrad\server.py", line 262, in handleRequest
    result = yield setting.handleRequest(self, c.data, data)
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1097, in _inlineCallbacks
    result = result.throwExceptionIntoGenerator(g)
  File "C:\Python27\lib\site-packages\twisted\python\failure.py", line 389, in throwExceptionIntoGenerator
    return g.throw(self.type, self.value, self.tb)
  File "ghz_fpga_server.py", line 1476, in run_sequence
    timingOrder)
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1099, in _inlineCallbacks
    result = g.send(result)
  File "ghz_fpga_server.py", line 743, in run
    raise TimeoutError(self.timeoutReport(runners, results))
  File "ghz_fpga_server.py", line 877, in timeoutReport
    if runner.executionCount is None:
exceptions.AttributeError: 'DacRunner_Build15' object has no attribute 'executionCount'

	at org.labrad.RequestReceiver.get(RequestReceiver.java:111)
	at org.labrad.RequestReceiver.get(RequestReceiver.java:41)
	at org.labrad.ServerConnection.sendAndWait(ServerConnection.java:526)
	at org.labrad.qubits.QubitContext.run_experiment(QubitContext.java:1106)
	at sun.reflect.GeneratedMethodAccessor18.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.labrad.handlers.SingleArgVoidHandler.handle(SingleArgVoidHandler.java:24)
	at org.labrad.ContextManager.processRequest(ContextManager.java:143)
	at org.labrad.ContextManager.access$100(ContextManager.java:15)
	at org.labrad.ContextManager$RequestProcessor.run(ContextManager.java:181)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.labrad.errors.LabradException: [GHz FPGAs] Remote Traceback (most recent call last):
  File "C:\pylabrad\labrad\server.py", line 262, in handleRequest
    result = yield setting.handleRequest(self, c.data, data)
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1097, in _inlineCallbacks
    result = result.throwExceptionIntoGenerator(g)
  File "C:\Python27\lib\site-packages\twisted\python\failure.py", line 389, in throwExceptionIntoGenerator
    return g.throw(self.type, self.value, self.tb)
  File "ghz_fpga_server.py", line 1476, in run_sequence
    timingOrder)
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1099, in _inlineCallbacks
    result = g.send(result)
  File "ghz_fpga_server.py", line 743, in run
    raise TimeoutError(self.timeoutReport(runners, results))
  File "ghz_fpga_server.py", line 877, in timeoutReport
    if runner.executionCount is None:
exceptions.AttributeError: 'DacRunner_Build15' object has no attribute 'executionCount'

	at org.labrad.RequestReceiver.set(RequestReceiver.java:171)
	at org.labrad.RequestDispatcher.finishRequest(RequestDispatcher.java:89)
	at org.labrad.ServerConnection.handlePacket(ServerConnection.java:587)
	at org.labrad.ServerConnection.access$100(ServerConnection.java:77)
	at org.labrad.ServerConnection$1.run(ServerConnection.java:314)
	... 1 more
 [payload=None]
```

pomalley, 2015-06-25:
Again:

```
Error: (0) java.util.concurrent.ExecutionException: org.labrad.errors.LabradException: [GHz FPGAs] Remote Traceback (most recent call last):
  File "C:\pylabrad\labrad\server.py", line 262, in handleRequest
    result = yield setting.handleRequest(self, c.data, data)
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1097, in _inlineCallbacks
    result = result.throwExceptionIntoGenerator(g)
  File "C:\Python27\lib\site-packages\twisted\python\failure.py", line 389, in throwExceptionIntoGenerator
    return g.throw(self.type, self.value, self.tb)
  File "ghz_fpga_server.py", line 1476, in run_sequence
    timingOrder)
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1099, in _inlineCallbacks
    result = g.send(result)
  File "ghz_fpga_server.py", line 743, in run
    raise TimeoutError(self.timeoutReport(runners, results))
  File "ghz_fpga_server.py", line 877, in timeoutReport
    if runner.executionCount is None:
exceptions.AttributeError: 'AdcRunner_Build7' object has no attribute 'executionCount'

	at org.labrad.RequestReceiver.get(RequestReceiver.java:111)
	at org.labrad.RequestReceiver.get(RequestReceiver.java:41)
	at org.labrad.ServerConnection.sendAndWait(ServerConnection.java:526)
	at org.labrad.qubits.QubitContext.run_experiment(QubitContext.java:1106)
	at sun.reflect.GeneratedMethodAccessor18.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.labrad.handlers.SingleArgVoidHandler.handle(SingleArgVoidHandler.java:24)
	at org.labrad.ContextManager.processRequest(ContextManager.java:143)
	at org.labrad.ContextManager.access$100(ContextManager.java:15)
	at org.labrad.ContextManager$RequestProcessor.run(ContextManager.java:181)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: org.labrad.errors.LabradException: [GHz FPGAs] Remote Traceback (most recent call last):
  File "C:\pylabrad\labrad\server.py", line 262, in handleRequest
    result = yield setting.handleRequest(self, c.data, data)
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1097, in _inlineCallbacks
    result = result.throwExceptionIntoGenerator(g)
  File "C:\Python27\lib\site-packages\twisted\python\failure.py", line 389, in throwExceptionIntoGenerator
    return g.throw(self.type, self.value, self.tb)
  File "ghz_fpga_server.py", line 1476, in run_sequence
    timingOrder)
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1099, in _inlineCallbacks
    result = g.send(result)
  File "ghz_fpga_server.py", line 743, in run
    raise TimeoutError(self.timeoutReport(runners, results))
  File "ghz_fpga_server.py", line 877, in timeoutReport
    if runner.executionCount is None:
exceptions.AttributeError: 'AdcRunner_Build7' object has no attribute 'executionCount'

	at org.labrad.RequestReceiver.set(RequestReceiver.java:171)
	at org.labrad.RequestDispatcher.finishRequest(RequestDispatcher.java:89)
	at org.labrad.ServerConnection.handlePacket(ServerConnection.java:587)
	at org.labrad.ServerConnection.access$100(ServerConnection.java:77)
	at org.labrad.ServerConnection$1.run(ServerConnection.java:314)
	... 1 more
 [payload=None]
```

DanielSank, 2015-06-25:
I really *really* thought we fixed this. It's something about `executionCount` should be `executionCounter`...

jwenner, 2015-06-26:
I know there was a related change in #135, #136. Maybe this helps?