labrad/servers #195: Data vault multihead doesn't recognize when a computer reboots

pomalley, 2015-06-26:
Redheatdos sometimes reboots under mysterious circumstances (BSOD? windows update?) and when it does sometimes (usually?) the data vault doesn't properly reconnect. It seems to think it is still connected (judging by `get_servers`) and only works when I `kick_managers` and then add it again.

DanielSank, 2015-06-26:
Why are we still running windows?

pomalley, 2015-06-26:
The old answer was the DE server, but now that we have the Scala DE server...

DanielSank, 2015-06-26:
Right. Let's move our lives into the happy land of OS's that work.

maffoo, 2015-06-26:
To be fair to windows, it seems like there are hardware issues on this box; none of our other machines have these problems. Not arguing against moving to linux, but I doubt putting linux on here will fix things.

DanielSank, 2015-06-26:
Ah. In that case

ejeffrey, 2015-06-27:
The datavault is supposed to have a keep-alive feature.  However, this is
contingent on the computer coming back up with the same IP.  Is redheat dos
listed on skynet's DHCP static assignment table?

Evan

ejeffrey, 2015-06-27:
Also, we can probably do a better job with the keepalives.  Right now it
only detects a connection lost when the remote host sends an RST packet in
response to data, which only happens if the OS is functioning, but the
connection is invalid (such as after a power failure/blue screen +
reboot).  It is also possible that the windows firewall is blocking these
connection lost packets -- we should probably decrease the firewall
agressivness on manager computers, the standard settings are not really
suitable for servers.  Regardless, we can add real timeout where if we
don't get something back to a ping in 60 seconds, we just assume the
connection is lost and drop it.

Evan