labrad/servers #209: TimeTrace testing

Brian-Ballard, 2015-08-06:
When trying to run timetrace.py
'''python
def get_time_trace(board_name):
    """Get a time trace from an ADC board

    Args:
        board_name (str): Name of the ADC from which you want to geta time
            trace. Check available boards by calling list_devices() on the fpga
            server.

    Returns:
        Tuple (I (ndarray(int)), Q (ndarray(int))).
    """
	
	
    with labrad.connect() as cxn:
        fpga = cxn.ghz_fpgas
        fpga.select_device(board_name)
        data = fpga.adc_run_average()
        return dataget
,,,

I can import labrad in my python console, but when i run timetrace.py i get the following dump:

In [89]: timetrace.get_time_trace('abg ADC 1')
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-89-4df77d1f3411> in <module>()
----> 1 timetrace.get_time_trace('abg ADC 1')

C:\users\brian\desktop\ADC_Bringup\timetrace.py in get_time_trace(board_name)
    10         Tuple (I (ndarray(int)), Q (ndarray(int))).
    11     """
---> 12         import labrad
    13
    14     with labrad.connect() as cxn:

NameError: global name 'labrad' is not defined

When i run all of the included functions from the console including import lab rad there is a problem with the arguments for the average function as below

In [1]: import labrad

In [2]: cxn = labrad.connect()

In [3]: fpga = cxn.ghz_fpgas

In [4]: fpga.select_device('abg ADC 1')
Out[4]: 'abg ADC 1'

In [5]: data = fpga.adc
fpga.adc_bringup           fpga.adc_register_readback
fpga.adc_demod_phase       fpga.adc_run_average
fpga.adc_demod_range       fpga.adc_run_calibrate
fpga.adc_filter_func       fpga.adc_run_demod
fpga.adc_mixer_table       fpga.adc_run_mode
fpga.adc_monitor_outputs   fpga.adc_trig_magnitude
fpga.adc_recalibrate       fpga.adc_trigger_table

In [5]: data = fpga.adc_run_average()
---------------------------------------------------------------------------
Error                                     Traceback (most recent call last)
<ipython-input-5-5013ba07d162> in <module>()
----> 1 data = fpga.adc_run_average()

C:\Users\Brian\Documents\GitHub\pylabrad\labrad\client.pyc in __call__(self, *ar
gs, **kw)
     63         if wrap:
     64             future.addCallback(lambda resp: resp[0][1])
---> 65         return future.wait() if wait else future
     66
     67     # data to be loaded on demand

C:\Users\Brian\Documents\GitHub\pylabrad\labrad\backend.pyc in wait(self)
    396                 # If any Future in the queue fails,
    397                 # we immediately bail.
--> 398                 result.raiseException()
    399             else:
    400                 for func, args, kw in f.callbacks:

C:\Users\Brian\Anaconda\lib\site-packages\twisted\python\failure.pyc in raiseExc
eption(self)

Error: (0) [GHz FPGAs] Remote Traceback (most recent call last):
  File "C:\Users\Brian\Anaconda\lib\site-packages\twisted\internet\defer.py", li
ne 1128, in _inlineCallbacks
    result = g.send(result)
  File "ghz_fpga_server.py", line 2130, in adc_run_average
    ans = yield dev.runAverage()
exceptions.TypeError: runAverage() takes exactly 5 arguments (1 given)
 [payload=None]

DanielSank, 2015-08-06:
1. add `import labrad` to the top of the script and reload it.

1. The problem with `run_average` looks like a bug in the server. Will check.

DanielSank, 2015-08-06:
By the way @Brian-Ballard to type in code use backticks (`) instead of single quotes.

maffoo, 2015-08-06:
what version of the ghz_fpga_server are you running?

DanielSank, 2015-08-06:
@maffoo LOL I should have asked that. You'd think I'd have learned after yesterday's adventure with the registry editor :-)

DanielSank, 2015-08-06:
@Brian-Ballard I'd recommend updating your git repository to the latest from our github repository. Let me know once you've done that.

Brian-Ballard, 2015-08-06:
updated.

DanielSank, 2015-08-06:
...and?

Brian-Ballard, 2015-08-06:
the import labrad added to the script give and unexpected indentation error 
```python
In [3]: import timetrace
  File "timetrace.py", line 13
    import labrad
    ^
IndentationError: unexpected indent
```

Brian-Ballard, 2015-08-06:
Same result from running in the shell.

maffoo, 2015-08-06:
it's python, so you need to get the indentation right. Presumably this is at the top of the script, so you can't have any whitespace before the import.

DanielSank, 2015-08-06:
@Brian-Ballard Ok, fix the indentation :-)

I can respond to these problems much more effectively you post the code you're trying to run. Since I can't see your computer screen I cannot debug an indentation error without seeing the code *exactly* as it is in your text editor.

Indentation errors happen if code is not consistently indented. I'd also strongly recommend using a text editor where you can see white space characters, and *never* use tabs in python code. Configure your editor to use 4 spaces instead of tabs.

Finally, I see you put the import statement on line 13. Always put import statements at the very top of the file.

Brian-Ballard, 2015-08-06:
Ok thanks for the python tips. Timetrace script now works to the same point as running the commands.  Still getting same runaAverage arguements dump:
```python
In [4]: timetrace.get_time_trace('abg ADC 1')
---------------------------------------------------------------------------
Error                                     Traceback (most recent call last)
<ipython-input-4-4df77d1f3411> in <module>()
----> 1 timetrace.get_time_trace('abg ADC 1')

C:\Users\Brian\Desktop\ADC_Bringup\timetrace.py in get_time_trace(board_name)
     16         fpga = cxn.ghz_fpgas
     17         fpga.select_device(board_name)
---> 18         data = fpga.adc_run_average()
     19         return dataget

C:\Users\Brian\Documents\GitHub\pylabrad\labrad\client.pyc in __call__(self, *ar
gs, **kw)
     63         if wrap:
     64             future.addCallback(lambda resp: resp[0][1])
---> 65         return future.wait() if wait else future
     66
     67     # data to be loaded on demand

C:\Users\Brian\Documents\GitHub\pylabrad\labrad\backend.pyc in wait(self)
    396                 # If any Future in the queue fails,
    397                 # we immediately bail.
--> 398                 result.raiseException()
    399             else:
    400                 for func, args, kw in f.callbacks:

C:\Users\Brian\Anaconda\lib\site-packages\twisted\python\failure.pyc in raiseExc
eption(self)

Error: (0) [GHz FPGAs] Remote Traceback (most recent call last):
  File "C:\Users\Brian\Anaconda\lib\site-packages\twisted\internet\defer.py", li
ne 1128, in _inlineCallbacks
    result = g.send(result)
  File "ghz_fpga_server.py", line 2130, in adc_run_average
    ans = yield dev.runAverage()
exceptions.TypeError: runAverage() takes exactly 5 arguments (1 given)
 [payload=None]
```

maffoo, 2015-08-06:
Looking at `adc.py`, I see that `runAverage` has a different type signature for adc build 1 (https://github.com/martinisgroup/servers/blob/master/GHzDACs/Cleanup/adc.py#L439) versus adc build 7 (https://github.com/martinisgroup/servers/blob/master/GHzDACs/Cleanup/adc.py#L972). Presumably your board is getting detected as a build below 7 and that code is being used, which is why the failure is happening.

Leaving aside for the moment the fact that this code is totally broken, do you know what build of the adc you are using?

Brian-Ballard, 2015-08-06:
I only have up to build 2 which is what I am using.

Brian-Ballard, 2015-08-06:
It works with the updated firmware.

DanielSank, 2015-08-06:
@maffoo would you I gave the server problem it's own issue, so closing this one.

closed 2015-08-06 by DanielSank