labrad/servers #214: No device error from ghz_fpga_server.py

changyunyoo, 2015-09-01:
Hi, 
  We're setting up AWG and stuck on the final board test where we are calling ghz_fpga_server.py. We get a "No devices detected on the board group" error and the list_devices() python command returns an empty array (Screenshot #1). We made sure that we put correct strings and port number in the register keys (Screenshot #2) and we set all the dip switches off such that the board number is 0 (dip switches pushed toward the edge of the board). We also tried changing the last two digits of the MAC addresses to match the board number but the same error still remained. So far, we tried two different sof files (GHzDAC_V9_B15_P30_P45_s4.sof and GHz_V8_B14_P22_P315_s15.sof). 
Any obvious ideas on why we can't detect the device? 

[image]

[image]

[image]

DanielSank, 2015-09-02:
The fpga server is pretty clearly saying that there's a timeout. When in the process does that happen?

changyunyoo, 2015-09-02:
The timeout errors occur when calling DeferredList method in the line answer = yield defer.DeferredList(detections, consumeErrors=True) in line #1013. But I thought there might be something we obviously missed since we can detect the device for our existing AWG even when we get the same timeout errors.

DanielSank, 2015-09-03:
>  we can detect the device for our existing AWG even when we get the same timeout errors.

1. What do you mean "existing AWG"?

1. What device are you trying to bring up?

DanielSank, 2015-09-03:
1. Have you used your setup to successfully talk to any boards in the past?

1. If not, what happened to the old setup that worked?

1. If you have talked to boards before, what is different about this setup? Can you talk to previously vetted boards with the present setup?

changyunyoo, 2015-09-03:
Sorry for the confusion. 

  1. We have a working AWG that uses the DAC board. We're now working on the second one. 
  2. It's the DAC board. 

  1. No, the old AWG is run by a different computer. The current setup has not detected any board yet. 
  2. We'e setting up the new AWG in a different room, and the old setup is still connected to the old DAC. 
  3. Now that I think about it, I can go check if the old setup can detect the new board and start tracking down the sources that cause the problem. But, as far as I know, the setups are identical except for the host name, port number, etc. 

Thanks !

changyunyoo, 2015-09-08:
It turned out that the security software in the new setup had been preventing the connection with the board. Turning off the protection solved the problem.

DanielSank, 2015-09-08:
Glad you guys got it all figured out.

Marking as closed. Reopen if the problem persists.

closed 2015-09-08 by DanielSank