labrad/servers #215: ghz fpga or direct ethernet server seems overwhelmed by high sampling rate.

btchiaro, 2015-09-04:
If I try to execute scans that are a single stat with 2x10^6 demods I sometimes get an ADC timeout error. I've run into this problem a couple of times, it seems to happen when there is more than one such scan in the pipeline or when someone else is also taking data.  Restarting the ghz fpga server and the direct ethernet server has fixed this in the past.  The traceback is below.

```
Error                                     Traceback (most recent call last)
<ipython-input-1-63a6559ec5ea> in <module>()
----> 1 reload(fb);fb.runFreshRapidBatch(s)

/home/ben/devBen/freshBatch.pyc in runFreshRapidBatch(s, points)
     40         for holdTime in holdTimes:
     41             holdTime = holdTime*U.ns
---> 42             frrto.runFreshRapidRTO(s,['q1'], holdTime=holdTime,  points=points, name=None,bonusname=None,bonusprename = None, save=True, collect=False,  noisy=False)
     43
     44 def runSlowFreshHoldTimeBatch(s,points = 1000000):

/home/ben/devBen/fresh_rapid_daq.pyc in runFreshRapidRTO(Sample, measure, holdTime, points, name, bonusname, bonusprename, save, collect, noisy, **kwargs)
     70         data_out = np.vstack((time_floats,states)).T
     71         returnValue(data_out)
---> 72     data = sweeps.grid(func, axes, dataset=dataset, save=True, noisy=noisy, **kwargs)
     73     if collect:
     74         return data

/home/ben/pyleBen/pyle/pyle/dataking/sweeps.pyc in grid(func, axes, dataset, **kw)
    320         dataset=dataset,
    321         abortPrefix=[1],
--> 322         **kw
    323     )
    324

/home/ben/pyleBen/pyle/pyle/dataking/sweeps.pyc in _run(func, sweep, save, dataset, abortable, abortPrefix, collect, noisy, pipesize)
    219         params = dataset.params
    220         name = dataset.name
--> 221         data = np.vstack(it)
    222         kwargs = {'number': dataset.num, 'path': dataset.path} if save else {}
    223         return pyle.datavault.LocalDataSet(data, independents, dependents,

/usr/local/lib/python2.7/dist-packages/numpy/core/shape_base.pyc in vstack(tup)
    226
    227     """
--> 228     return _nx.concatenate([atleast_2d(_m) for _m in tup], 0)
    229
    230 def hstack(tup):

/home/ben/pyleBen/pyle/pyle/datasaver.pyc in capture(self, iterable)
    146         """
    147         with self as dataset:
--> 148             for data in iterable:
    149                 dataset.add(data)
    150                 yield data

/home/ben/pyleBen/pyle/pyle/pipeline.pyc in gen()
     94                 runner = _FutureRunner(result)
     95             elif is_generator(result):
---> 96                 runner = _GeneratorRunner(result)
     97             else:
     98                 runner = _ValueRunner(result)

/home/ben/pyleBen/pyle/pyle/pipeline.pyc in __init__(self, gen)
    118         self.gen = _run(gen)
    119         self.done = False
--> 120         self.advance()
    121
    122     def advance(self):

/home/ben/pyleBen/pyle/pyle/pipeline.pyc in advance(self)
    123         if not self.done:
    124             try:
--> 125                 self.gen.next()
    126             except _DefGen_Return, e:
    127                 self.done = True

[... 91 lines cut ...]
Vince DAC 15: OK Expected executions: 1  Actual: 0
Vince DAC 28: OK Expected executions: 1  Actual: 0
Vince DAC 18: OK Expected executions: 1  Actual: 0
Vince DAC 26: OK Expected executions: 1  Actual: 0
Vince DAC 14: OK Expected executions: 1  Actual: 0
Vince DAC 11: OK Expected executions: 1  Actual: 0
Vince DAC 24: OK Expected executions: 1  Actual: 0
Vince ADC 1: timeout! Expected executions: 1  Actual: 1
Vince DAC 16: OK Expected executions: 1  Actual: 0
Vince DAC 29: OK Expected executions: 1  Actual: 0
Vince DAC 13: OK Expected executions: 1  Actual: 0
Vince DAC 19: OK Expected executions: 1  Actual: 0
Vince DAC 12: OK Expected executions: 1  Actual: 0
Vince DAC 25: OK Expected executions: 1  Actual: 0
Vince DAC 27: OK Expected executions: 1  Actual: 0
Vince DAC 17: OK Expected executions: 1  Actual: 0

    at org.labrad.RequestReceiver.set(RequestReceiver.java:171)
    at org.labrad.RequestDispatcher.finishRequest(RequestDispatcher.java:89)
    at org.labrad.ServerConnection.handlePacket(ServerConnection.java:587)
    at org.labrad.ServerConnection.access$100(ServerConnection.java:77)
    at org.labrad.ServerConnection$1.run(ServerConnection.java:314)
    ... 1 more
 [payload=None]
```

btchiaro, 2015-11-18:
I'm continuing to have this problem.  Previously, I had reduced the number of demods in my scan to 10^6 and that seemed to fix the problem.  Not that Vince is being run by t1000, this issue is appearing more often.  @rb25 Mentioned that he had a similar timeout issue today.  I'm the code below is called from the add_rapidRTO branch of pyle.
```
ben@bcpc:~$ ipython
Python 2.7.6 (default, Jun 22 2015, 17:58:13)
Type "copyright", "credits" or "license" for more information.

IPython 3.1.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.
:0: FutureWarning: IPython widgets are experimental and may change in the future.
Registry Root is ['', 'Ben']
Sample Path is ['qubit', 'ManHatMon_90nm', '151117_q4_5.492GHz']
Data Vault directory is ['', 'Ben', 'qubit', 'ManHatMon_90nm', '151117_q4_5.492GHz']

In [1]: import pyle.dataking.rapidRTO_acquisition as rra

In [2]: rra.runRapidRTO(s,'q4')
---------------------------------------------------------------------------
Error                                     Traceback (most recent call last)
<ipython-input-2-cc36ed810818> in <module>()
----> 1 rra.runRapidRTO(s,'q4')

/home/ben/pyleBen/pyle/pyle/dataking/rapidRTO_acquisition.pyc in runRapidRTO(Sample, measure, holdTime, samplingInterval, points, Phases, name, bonusname, bonusprename, save, collect, noisy, **kwargs)
    149
    150     data = sweeps.grid(func, axes, dataset=dataset,
--> 151             save=True, noisy=noisy, **kwargs)
    152
    153     if collect:

/home/ben/pyleBen/pyle/pyle/dataking/sweeps.pyc in grid(func, axes, dataset, **kw)
    320         dataset=dataset,
    321         abortPrefix=[1],
--> 322         **kw
    323     )
    324

/home/ben/pyleBen/pyle/pyle/dataking/sweeps.pyc in _run(func, sweep, save, dataset, abortable, abortPrefix, collect, noisy, pipesize)
    219         params = dataset.params
    220         name = dataset.name
--> 221         data = np.vstack(it)
    222         kwargs = {'number': dataset.num, 'path': dataset.path} if save else {}
    223         return pyle.datavault.LocalDataSet(data, independents, dependents,

/usr/local/lib/python2.7/dist-packages/numpy/core/shape_base.pyc in vstack(tup)
    226
    227     """
--> 228     return _nx.concatenate([atleast_2d(_m) for _m in tup], 0)
    229
    230 def hstack(tup):

/home/ben/pyleBen/pyle/pyle/datasaver.pyc in capture(self, iterable)
    146         """
    147         with self as dataset:
--> 148             for data in iterable:
    149                 dataset.add(data)
    150                 yield data

/home/ben/pyleBen/pyle/pyle/pipeline.pyc in gen()
     94                 runner = _FutureRunner(result)
     95             elif is_generator(result):
---> 96                 runner = _GeneratorRunner(result)
     97             else:
     98                 runner = _ValueRunner(result)

/home/ben/pyleBen/pyle/pyle/pipeline.pyc in __init__(self, gen)
    118         self.gen = _run(gen)
    119         self.done = False
--> 120         self.advance()
[... 90 lines cut ...]
  File "C:\Python27\lib\site-packages\twisted\internet\defer.py", line 1107, in _inlineCallbacks
    result = g.send(result)
  File "ghz_fpga_server.py", line 747, in run
    raise TimeoutError(self.timeoutReport(runners, results))
__main__.TimeoutError: Some boards failed:
Vince DAC 12: OK Expected executions: 1  Actual: 0
Vince DAC 26: OK Expected executions: 1  Actual: 0
Vince DAC 24: OK Expected executions: 1  Actual: 0
Vince DAC 13: OK Expected executions: 1  Actual: 0
Vince DAC 14: OK Expected executions: 1  Actual: 0
Vince DAC 17: OK Expected executions: 1  Actual: 0
Vince ADC 1: timeout! Expected executions: 1  Actual: 1
Vince DAC 11: OK Expected executions: 1  Actual: 0
Vince DAC 16: OK Expected executions: 1  Actual: 0
Vince DAC 27: OK Expected executions: 1  Actual: 0

        at org.labrad.RequestReceiver.set(RequestReceiver.java:171)
        at org.labrad.RequestDispatcher.finishRequest(RequestDispatcher.java:89)
        at org.labrad.ServerConnection.handlePacket(ServerConnection.java:587)
        at org.labrad.ServerConnection.access$100(ServerConnection.java:77)
        at org.labrad.ServerConnection$1.run(ServerConnection.java:314)
        ... 1 more
 [payload=None]

```

maffoo, 2015-12-03:
I think we've found and fixed the underlying issue: martinisgroup/servers#279. Interested parties are encouraged to go review that :)

maffoo, 2015-12-03:
@btchiaro, not sure how much you've been seeing this lately, but it's likely due to the issue fixed by #279. I'm going to close this for now, but please reopen if you continue to have problems after updating to the latest ghz_fpga_server version.

closed 2015-12-03 by maffoo