labrad/servers #221: Need build and install script

ejeffrey, 2015-09-25:
In the past, this repo was a collection of 'runnable' servers.  Most of them were python, and the remaining Java or Delphi servers were stored as executables. This allowed you to just point a node at it and go.  Now that we are trying to have only source here, we need a way to automatically make an installer that puts all the servers in one place for the node.

DanielSank, 2015-09-25:
Can you give an example of a server that doesn't just work? I guess you're having problems with the qubit server, is that right?

ejeffrey, 2015-09-25:
Neither the qubit server nor the direct ethernet server are runnable as is.  I got the direct ethernet server to build earlier, but when I try it now I appear to get something that isn't runnable.  The qubit server doesn't even compile.  If the servers repo isn't going to be a place we go to deploy a suite of ready-to-go labrad servers we need that.

maffoo, 2015-09-25:
The qubit server compiles fine for me if I cd into the QubitServer directory and run `sbt compile`. To actually produce a package, run `sbt packArchive` and it will produce a .tar.gz in the `target/` directory. Just unpack this archive on the node machine in a directory visible to the node. If the compile is failing for you, can you post some of the error output so we can diagnose?

The steps for the direct ethernet server are the same because its sbt project has been set up the same way. Do you have any information about how it fails to start?

ejeffrey, 2015-09-25:
OK, I was using sbt package (as per the online sbt docs), which generates a
.jar file, but the jar file doesn't run.  packArchive generates the
.tar.gz.  You sent me that command earlier, but hangouts expired it.  Is
there a way we can just make a runanble jar?  It seems strange to have a
.tar.gz file with a bunch of .jar files inside it.

The QubitServer was failing because I had the java8 JRE but an older JDK.
Can we get a version error if we try to use the wrong compiler?

maffoo, 2015-09-25:
By default, the jar that sbt produces does not contain any of the dependencies, which is why it won't run unless you add all the dependency jars on the classpath (see the ini files in either of these server projects for the command line to do that). It's possible to repackage everything into a single runnable jar, but I've had issues with that approach in the past and we abandoned it at nest in favor of leaving the dependencies exploded like this. The installation here just requires copying the archive and then unpacking, which seems like it's not much worse than copying a single jar file; is there a particular reason you'd prefer the latter?

As for the compiler issue, I'm guessing it was the reverse, that you compiled with JDK 8 and then tried to run on JRE 7? Classes compiled with JDK 7 should run fine on java 8.

ejeffrey, 2015-09-25:
It was javac from jdk6, which just won't compile due to syntax errors.

Evan

ejeffrey, 2015-09-25:
To be clear, the documentation for building scala packages is great, but we need a script to make a ready-to-go directory of servers that you can point a node at, and which we reject PRs that don't have that.  Otherwise next time someone uses a new language or build system, we are going to go through this again.